mne-tools/mne-python #2245: Q: XDawn implementation

alexandrebarachant, 2015-06-27:
Hi,
I have python code for XDawn algorithm, and I though it will be interesting to push it on MNE.
However, i'm not sure of the API and where to put it.

Xdawn algorithm is two part. The first one is a least square estimation of the evoked responses, taking into account the possible overlap of ERPs. It could be a standalone method, something like : 

`evokeds = least_square_evoked(raw, events, event_id, tmin, tmax, decim, picks)`

this method would return a list of evoked objects, one for each type in event_id.

The second part is a generalized Rayleigh quotient to estimates a set of spatial filters that maximize the signal to signal + Noise ratio.

The Xdawn api could be something similar to the ICA Instance 

`xd = Xdawn(n_components=4,...)`

with a fit method

`xd.fit(raw, events, event_id, tmin, tmax, decim, picks)`

then, the Xdawn instance could have some methods like `plot_components()`, `get_sources()`, `plot_source_evoked()`, etc.

Where is the best place to put this ? preprocessing maybe ?
What do you think ?

agramfort, 2015-06-27:
> Xdawn algorithm is two part. The first one is a least square estimation of
> the evoked responses, taking into account the possible overlap of ERPs. It
> could be a standalone method, something like : [...]
do you really need to start from raw? why not epochs? I am just afraid of
duplicating epochs construction params here.

> The second part is a generalized Rayleigh quotient to estimates a set of
> spatial filters that maximize the signal to signal + Noise ratio.
>
> The Xdawn api could be something similar to the ICA [...]
same rmk why not starting from epochs?

> Where is the best place to put this ? preprocessing maybe ?
> What do you think ?
>
yes preprocessing.

once you have xd what do you do with it? you get projs?

how would a full decoding or simple evoked denoising pipeline look like?

alexandrebarachant, 2015-06-27:
> do you really need to start from raw? why not epochs? I am just afraid of
duplicating epochs construction params here.

The algorithm works with continuous signal. in the case your ERPs are not overlapped, then you could simply use epochs, because it is equivalent to the standard average.
It is technically possible to reconstruct pseudo-continuous signal from epochs and events position, and run the algorithm on it, but it is a source of bug IMO.

> same rmk why not starting from epochs?

same + spatial filters are computed from the ERP covariance and the signal covariance, which mean to have access to the full signal.

> once you have xd what do you do with it? you get projs?

For my specific case, i only use it to visualize spatial patterns and "sources" evoked potential.
i like this algorithm because it deal very well with overlapped responses and is very effective to reduce dimensionality. 

> how would a full decoding or simple evoked denoising pipeline look like?

From a decoding point of view, xd is used as a preprocessing before epoching and classification. However, since it need all the signal before epoching, it is not very compatible with the current decoding API. For this reason i made a simplified version that works with epoched signal, but which does not take into account the overlap in the responses (one of the main advantage of this method).

it could be also used for denoising, similarly to an ICA (you find the 2 components that maximize the evoked responses, nullify the others, and back-project the signal)

agramfort, 2015-06-28:
> The algorithm works with continuous signal. in the case your ERPs are not overlapped, then you could simply use epochs, because it is equivalent to the standard average.
> It is technically possible [...]

I would try to make it work from epochs as the API would be much
simpler and it would avoid some code duplication with epoching
mechanism but I can be wrong... Maybe I should look at the code to get
a better opinion.

> same rmk why not starting from epochs?
>
> same + spatial filters are computed from the ERP covariance and the signal covariance, which mean to have access to the full signal.

why not providing MNE covariance objects?

I am not trying to convince you but challenging your vision :)

> once you have xd what do you do with it? you get projs?
>
> For my specific case, i only use it to visualize spatial patterns and "sources" evoked potential.
> i like this algorithm because it deal very [...]

to me it sounds like a transformer in the sklearn API sense. Making it
work with epochs would make it very similar to a CSP transformer as we
have it already.

> it could be also used for denoising, similarly to an ICA (you find the 2 components that maximize the evoked responses, nullify the others, and back-project the signal)

I like that. I would vote for objects that implement fit_transform as
the CSP class.

alexandrebarachant, 2015-06-28:
> I would try to make it work from epochs as the API would be much
simpler and it would avoid some code duplication with epoching
mechanism but I can be wrong... Maybe I should look at the code to get
a better opinion.

i also don't like code and API duplication. The least square estimation could works from epochs. To do so i would have to reconstruct the original signal using the events position and merge part of the signal that are overlapped. En the End, it would only works if the information of epochs.events is in sync with the data. My concern is that it will be misleading to the user, thinking that it will works with **any** Epoch instance, while such methods like `concatenate_epochs`, `decim`, `resample` or `crop` will break the sync between events posision and the data.

> why not providing MNE covariance objects?

I will use mne covariance object. especially for the "noise" covariance. I made a first try for the implementation [here](https://github.com/alexandrebarachant/mne-python/blob/xdawn/mne/preprocessing/xdawn.py) but i got lost in the mne covariance code before being able to use it :(

However, i dont want the main method to only takes covariances as input. it will be too complicated for the user, and also require a deep understanding of the algoritm itself. My goal here is to provide an implementation as close as the description in the [paper](http://www.hal.inserm.fr/file/index/docid/619997/filename/2011_EUSIPCO.pdf). It should be practical to use without worrying about all the small subtlety of the algorithm.

> to me it sounds like a transformer in the sklearn API sense. Making it
work with epochs would make it very similar to a CSP transformer as we
have it already.

I already have a transformer in my pyriemann toolbox [here](https://github.com/alexandrebarachant/pyRiemann/blob/master/pyriemann/spatialfilters.py). It is a simplified version to be 100% compatible with the sklearn api. It is more or less equivalent to the original algorithm, but works only for non-overllaped ERP. It is also lighter and faster. We can integrate it if you want.

As i said, my goal is to provide the full original algo, which is interesting outside the decoding perspective. I use it to extract spatial patterns and un-convolved ERP average when i use an overlapped experiment desing. In this sense, it is much more close to the ICA than the CSP. One option is to add proj objects so i would be able to use the mne object related to them.

alexandrebarachant, 2015-06-28:
Hi again, 
I implemented the denoising part, working similarly to the ICA, but is supervised and built to increase the SSNR of the ERP.
it works pretty well.
for examples, here is the epochs image before denoising : 

[image]

and after : 

[image]

It works also on the raw data, where you can actually see the ERP : 
[image]

and a zoom out on the same part 

[image]

we can clearly see the ERP occurring at a constant pace.

wmvanvliet, 2015-06-29:
From a user's standpoint, I see no other reasonable option than to work with MNE's Epochs object.

With the plain and simple Epochs procedure, the original timeline can be reconstructed, since the event onsets are properly stored. 

Can we make `concatenate_epochs`, `decim`, and `resample` properly deal with event onsets? I see no real downside to that, since the event onset/duration columns become meaningless if we don't resample them too.

When constructing an EpochArray object, the user is free to construct the events in any way deemed appropriate, so no guarantees are given.

Can we add an `correct_overlap` flag to the constructor of the XDawn object? If epochs are non-overlapping, you don't want the large computational overhead of applying the correction. When the flag is set to true, a check can be performed to determine whether the event onsets/durations would actually produce overlapping Epochs, and other sanity checks, and give an error if this is not the case. In this case, the user can still use the XDawn algorithm even when the event markers have become unusable, in the full knowledge that overlap correction will not be performed.

alexandrebarachant, 2015-06-29:
> From a user's standpoint, I see no other reasonable option than to work with MNE's Epochs object.

Then, the user would have to provide the raw covariance in case he want to apply the algorithm in the best possible condition. but it is still acceptable.

> Can we make concatenate_epochs, decim, and resample properly deal with event onsets? I see no real downside to that, since the event onset/duration columns become meaningless if we don't resample them too.

Sure, it's more a bug than a true limitation. I just wanted to point this as a potential source of bug since most of the Epochs methods doesn't care about the event positions after epoching.
I can do a PR for this.

> Can we add an correct_overlap flag to the constructor of the XDawn object?

yep, we can add this flag, with a 'auto' / True / False  where auto select the best approach given the event positions.

Last thing, the algorithm train a set of spatial filter for each event_id, so when you call `apply`, it return a denoised version of the signal for each condition. does it seems ok to you ?

agramfort, 2015-06-29:
@alexandrebarachant can you write a code snippet a user would write? I like to think about API issues looking at code. It does not matter if the gist is not backup by actual code.

btw your preliminary plots are great !

alexandrebarachant, 2015-06-29:
with the epoch API, this can look like this [gist](https://gist.github.com/alexandrebarachant/b4d94245db162f6663f7)
if the signal covariance is not provided, then the aglo will use the covariance of the signal reconstructed from the epochs.

the main part of the code is done at 80%

wmvanvliet, 2015-06-29:
that looks like a good API to me.

alexandrebarachant, 2015-06-29:
the snipet works with my xdawn branch.
you can play with it if you want.
I will improve the code and the doc and open a PR

wmvanvliet, 2015-06-29:
> Last thing, the algorithm train a set of spatial filter for each event_id, so when you call apply, it return a denoised version of the signal for each condition.

This should be clearly documented. It confused me in the gist.

agramfort, 2015-06-30:
> with the epoch API, this can look like this gist
> if the signal covariance is not provided, then the aglo will use the covariance of the signal reconstructed from the epochs.
>
> the main part of the code is done at 80%

great. Looks like we have a plan !

looking forward to it.

dengemann, 2015-06-30:
@alexandrebarachant this is very exciting and I like the API a lot. What was the status of the Raw/Epochs discussion. I see no reason not to have both as possible inputs to fit, as is possible with our ICA class.

alexandrebarachant, 2015-06-30:
> This should be clearly documented. It confused me in the gist.

@wmvanvliet yes this is a bit confusing. the apply method takes event_id as a param, so you can specify which class of filter to apply. If not provided, all the filters are applied. In both case, the method return a dict of epoch instance. i don't know if there is a better option.

@agramfort I will have time to finish this later this week. I also want to add some method to visualize spatial patterns and the corresponding source evoked responses.

@dengemann I removed the fit from raw. I can put it back, but there is a lot of duplicated code with the epoch class, especially all the checks. Anyway, if you provide the signal covariance from raw, the results is strictly equivalent.

dengemann, 2015-06-30:
@alexandrebarachant for comprehension, which data should the signal covariance include conceptually for Xdawn? Only signal of interest or noise segments too or only noise / signal of non-interest segments?

alexandrebarachant, 2015-06-30:
@dengemann In the original paper, it is the covariance of all the signal (including signal of interest). The algorithm optimize the signal to (signal + noise) ratio of the ERP. A natural extension would be to provide the noise cov only, so that you will optimize the SNR.

alexandrebarachant, 2015-06-30:
@dengemann as a side note, Xdawn can be seen as a Whitening + SVD of the evoked response.

dengemann, 2015-06-30:
This is interesting! I will play with it as soon as I find some time. Might also be interesting to see how it interacts with cov-estimation options and to explore the baseline cov use case.

alexandrebarachant, 2015-06-30:
indeed. For the MEG data provided with MNE, i don't expect a very strong difference, since the ERP is very strong and will dominates the solution. For noisier data, providing a noise cov only may lead to improvements.

agramfort, 2015-06-30:
@alexandrebarachant can you open a PR with your xdawn branch so we can start to look?

jona-sassenhagen, 2015-07-15:
How similar is this to rERPs? See #2304

alexandrebarachant, 2015-07-15:
@jona-sassenhagen I just saw the PR. I think it is very similar, at least for the evoked response estimation. I have to read the paper, but i definitely think we could share some code.

alexandrebarachant, 2015-07-20:
Merged in #2296

closed 2015-07-20 by alexandrebarachant