mne-tools/mne-python #2255: All plots should have axes parameter

teonlamont, 2015-07-03:
I think that we should make sure all of our plotting functions have the `axes` parameter.

Most of them do but some don't, e.g. plot_projs_topomap. It would be great for added flexibility with creating figures.

cc: @mainakjas @jaeilepp

ImmanuelSamuel, 2015-07-07:
power.plot() needs axes parameter.

teonlamont, 2015-07-08:
related pr #2195

teonlamont, 2015-07-09:
evoked.plot mentioned in https://github.com/mne-tools/mne-python/pull/2241#issuecomment-116621986

teonlamont, 2015-07-12:
TFR.plot

jaeilepp, 2015-07-16:
``evoked.plot`` actually has an axes param.
``plot_projs_topomap`` creates multiple axes instances. Their positioning rely on the layout. I don't see the point in positioning them yourself.

teonlamont, 2015-07-16:
what about in the case of just one proj being plotted. that is the use case that I was thinking of. for instance, I want to make a figure that has the waveform with the proj applied on the left, and then the topography of that projection on the right.

jaeilepp, 2015-07-16:
Oh yeah. You're right. I must have looked at something else...

teonlamont, 2015-07-16:
you were right that if you have projs > 1, plot_projs_topomap will make subplots of all of the figures. I was thinking there should be an option when projs = 1, then you can use axes.

teonlamont, 2015-07-24:
closed for #2281

closed 2015-07-24 by teonlamont