mne-tools/mne-python #2258: find_layout doesn't work for KIT system if a channel is dropped

teonlamont, 2015-07-03:
Is there a way to preserve the original number of channels in a system after doing inst.pick_types. The two KIT systems differ in layout and they are distinguished based on the number of channels.

Eric89GXL, 2015-07-04:
Is there a more robust way too distinguish them, e.g. by coil type? Using
number of channels seems fragile. How does it work for Neuromag systems?

teonlamont, 2015-07-04:
unfortunately, they have the same coil type. I agree that using the number of channels is a weak assertion. Neuromag looks for particular channel types

agramfort, 2015-07-05:
can you hack something?

teonlamont, 2015-07-08:
I think the solution would be to do any layout getting before doing picking, e.g. `epochs.pick_types(meg=True, exclude='bads')`. since there is no distinguishing attribute between the two, that would be no elegant solution

closed 2015-07-08 by teonlamont