mne-tools/mne-python #2259: __repr__ for Epochs when read from disk

teonlamont, 2015-07-03:
the __repr__ of an Epochs objects read from disk is `_BaseRaw`. should this be `Epochs`? I guess we can't distinguish between Epochs and EpochsArray saved to disk and that is why the generic basename is being used.

cc: @Eric89GXL

Eric89GXL, 2015-07-04:
What we should really do is rename Epochs to something else (like
EpochsFromRaw but a less crappy a name) and change _BaseEpochs to Epochs,
but that will break people's code if they have subclassed. So I'm not sure
what a good solution is.

teonlamont, 2015-07-04:
sorry, that was supposed to saw `_BaseEpochs`, not `_BaseRaw`. but actually, nvm. I just loaded epochs from disk and it says `Epochs` now. I think I am ok with how things are now without any changes

teonlamont, 2015-07-09:
actually, i was wrong. i think it is still there.

Eric89GXL, 2015-07-09:
Yeah, it should still be there. I don't have a good solution in mind. We could just make `BaseEpochs` public.

teonlamont, 2015-07-09:
what do you think about doing this in the read_epochs:

```python
epochs = _BaseEpochs(info, data, events, event_id, tmin, tmax,
                             baseline, name=name, on_missing='ignore',
                             verbose=verbose)
epochs.__class__.__name__ = 'Epochs'
```

Eric89GXL, 2015-07-09:
BOOO HISSS monkey patch :) I think I'd actually prefer making `BaseEpochs` public instead.

teonlamont, 2015-07-09:
hahaha, but then what would make people use `EpochsArray` instead of `BaseEpochs`, they look similar? or simply, what's the use of making it public other than the repr?

Eric89GXL, 2015-07-09:
There isn't really a purpose. We can tell people they shouldn't instatiate it directly, and link to the other classes. To actually answer your question, EpochsArray has a few checks in it, otherwise it is a very thin wrapper around BaseEpochs.

agramfort, 2015-07-10:
I am lost. Can you paste a gist to reproduce the issue?

Eric89GXL, 2015-07-10:
When you read any `-epo.fif` file and look at the `__repr__`, the class name is `<_BaseEpochs : ...>`

agramfort, 2015-07-11:
see https://github.com/mne-tools/mne-python/pull/2272

PR #2272 (2015-07-11): FIX : BaseEpochs repr
fixes #2259
closed 2015-07-11 by teonlamont