mne-tools/mne-python #2260: STC iterator over time

teonlamont, 2015-07-03:
To save on memory and processing, it would be useful to have the stc generator compute over time. Currently it iterates over epochs. The use case is computing regressions in source space, or other parameter maps of source estimates over time.

Eric89GXL, 2015-07-04:
Would an iter_time method work for your use case?

dengemann, 2015-07-04:
@teonlamont FYI you can compute betas from a linear regression and map them to source space using MNE, it's valid cf. @olafhauk work.
I guees you want to do it for your ridge though and I bet it won't be valid for the ridge coefs.

teonlamont, 2015-07-04:
@Eric89GXL yeah, something like that is what I was thinking about.

there was a long discussion over the listserv a while back about using beta maps generated in sensor space then using them in source space https://mail.nmr.mgh.harvard.edu/pipermail//mne_analysis/2014-February/001841.html. @dengemann I agree that it should be ok for MNE, but what about for dSPM

Eric89GXL, 2015-07-04:
DSPM is linear too so I think it should be okay

teonlamont, 2015-07-04:
but what about using free orientation, I guess the transformation would be ok, but then the x, y, z values are combined

Eric89GXL, 2015-07-04:
Oh you mean loose or free orientation, yeah that combination of terms is
nonlinear

Eric89GXL, 2017-02-16:
@teonbrooks do we still need this?

teonbrooks, 2017-02-17:
I think we can close it for now. if anyone needs in the feature, we can reopen it. still think it could be useful

closed 2017-02-17 by teonbrooks