mne-tools/mne-python #2264: Document data mangling functionality

wmvanvliet, 2015-07-06:
This is one of the documentation area's I want to cover during the MNE sprint.
I would like to have a clean document that describes how to perform common data manipulations, such as:

 * **create** RawArray/EpochsArray/EvokedArray objects
 * **select** channels, time points, classes, etc.
 * **drop** channels, time points, classes, etc.
 * **concatenate** along channels, time points, classes, etc.
 *  **add** channels, time points, classes, etc.

Documenting these processes might have the side effect of raising a few API issues with the current `info` object.

`<rant>`

>Let's face it: many of our users (maybe even the majority) are researchers just like us who don't shy away from coding something. Researchers are usually doing things no-one has ever done before. The issues list is a nice example: so many are "hey, MNE should support this thing I want to do".

>But MNE can never support all the crazy things researchers want to do. Instead, we should aim to make it trivial to hack in your own crazy calculations and interface them with MNE to benefit from the wide range of things it does implement.

>At the core of this are the data structures of MNE. In order to make your code interface with the rest of MNE, it must be able to read its data structures and write the result as one. On the one hand, this means it should be trivial to create RawArray/EpochsArray/EvokedArray/STC objects. On the other hand, this means the programmer should be able to mould the existing MNE-objects like clay. In fact, a programmer should prefer to use our Raw/Epochs/Evoked/STC classes over raw Numpy array's, because they make manipulating the data easier and take care of the boring bookkeeping (making sure the channel names line up with the rows in the data array etc.).

>Examples of packages that get this right: Scikit-learn, Pandas

`</rant>`

dengemann, 2015-07-06:
Sounds good to me.

choldgraf, 2015-07-13:
@wmvanvliet are you working on this at the onset of the sprint? I can perhaps help out since I'm relatively new to MNE compared you all, and might have the perspective of a newbie which is probably useful for a tutorial. I'm out of Paris until Wednesday, however.

How do you imagine building these tutorials? As a part of the 'doc' folder, or as a separate tutorial (e.g., ipython notebook hosted on the MNE python github, or something like this)?

agramfort, 2015-07-13:
I am looking at how much

https://github.com/sphinx-gallery/sphinx-gallery/pull/36

can allow us to do the job.

wmvanvliet, 2015-07-13:
@choldgraf I would like to make it part of the 'doc' folder. The exact format is not terribly important. I guess I will write the text first using a notebook. Later, we can convert it into whatever documentation format we end up using.

I will work on this starting tomorrow at the latest, hopefully sooner.

choldgraf, 2015-07-13:
ok cool - do you want to make one big document or lots of little ones for
data slicing/merging/etc? Is it something that I can help with at the
start, or better to wait and help review once drafts are ready?

If it's helpful at all, I wrote two posts about MNE munging a little bit
ago:

http://chrisholdgraf.com/using-mne-python-to-extract-trials-and-epochs-from-data/
http://chrisholdgraf.com/using-mne-with-custom-or-non-standardized-data-formats/

Eric89GXL, 2015-07-13:
@choldgraf I just replied to the wrong thread, but I was saying maybe you
could work on viz tweaks and fixes

choldgraf, 2015-07-13:
hehe I just saw that - yes I can help with this as well as it seems like
people are already working on the docs/tutorials and on circular stats.
Don't have a ton of experience with viz libraries but I'm sure that I can
pick it up quickly. Is there a list of additions we'd like to add sitting
around somewhere? The main thing that would be relatively quick to do is
letting you interactively adjust the scaling, but I am sure there are other
things...

Eric89GXL, 2015-07-13:
I think scaling actually has been added already. Any related viz issues
I've tried to tie back to your open issue, so looking through that should
be sufficient. But Jakko has been quite productive so some of the changes
have probably been done already.

wmvanvliet, 2015-07-13:
@choldgraf lets discuss this on Gitter, I've send you an invite.

choldgraf, 2015-07-13:
OK cool, I am out for a little bit but will ping you when I have access to
good Internet. I'll be spotty on connectivity today and tomorrow just FYI.

Eric89GXL, 2017-02-16:
I think we have tutorials covering this now, but feel free to reopen if I'm wrong

closed 2017-02-16 by Eric89GXL