mne-tools/mne-python #2268: ROI definition in picks - ENH suggestion

ImmanuelSamuel, 2015-07-08:
For the plot_topo functions it would be nice if the picks argument can take multiple channel groups Eg: picks = [[1,2,3],[4,5,6]]. The plots can be shown at the centroid of the channels within a group. This can reduce the need to plot all channels resulting in really small plots for data recorded with a high density channel system.

wmvanvliet, 2015-07-09:
`plot_topo` uses a `Layout` object that defines the drawing position of the channels. The simplest way to achieve what you describe is to write your own layout file. Usually, you use the same recording setup over and over again, so investing an hour or so in creating your own custom layout is worth it.

If there is a pressing need to be able to do this in code, I suggest adding it to the `make_eeg_layout` or `generate_2d_layout` functions.

teonlamont, 2015-07-10:
maybe the Layout object can generate default ROI layouts for the sensor grid

jona-sassenhagen, 2015-07-17:
@teonlamont remember the code for plotting ROIs I sent you some time ago?

teonlamont, 2015-07-17:
yeah, I have it around somewhere. my comment was inspired by you @jona-sassenhagen :)

PR #5071 (2018-04-27): MRG: Evoked stat example
I'm proposing an example for ERP ROI plotting (a topic which comes up ever so often). It also shows how to run and display typical stats on ERP data.

Arguably,
Closes #2268

Better than #2602
Solves what remains of #2693
closed 2018-04-27 by agramfort