mne-tools/mne-python #2270: Regression for spatiotemporal cluster tests

teonlamont, 2015-07-09:
Our regression function isn't compatible with the spatiotemporal cluster tests. We give a `namedtuple` with t-values, p-values, and beta-values as an output whereas functions like `scipy.stats.f_oneway` outputs the F-value and the p-value. also, we need to supply a design matrix for the regression and there isn't a way to pass attributes to the stat function. 
maybe it would be useful to have the ability to submit the stat and p-values as an argument, instead of the data and the statistical test?

Thoughts:
cc: @dengemann

dengemann, 2015-07-09:
I'd try a regular ttest or just a function that returns the data as stamt fun and pass the betas. You don't need to compute the regression during permutations. Think of your betamaps as weighted average. You also don't recompute you ERPs after peemuting.

teonlamont, 2015-07-10:
don't you need to recompute the regression when you shuffle the labels for the permutation? that's the basis of the distribution you generating, right?

dengemann, 2015-07-10:
Fot the t-test variant of our permutation clustering you assume a paired contrast and build the distribution by permuting the signs as those reflect differences between conditions. That should be the case too for your betas, not?
How does your regression problem look like?

teonlamont, 2015-07-10:
what I would like to do and my understanding of how this would be best implemented is that I fit a regression (design matrix with my predictor) to my MEG data. using a threshold (p <= .05 or t >= 2), I would calculate the largest space-time cluster that survives thresholding. for the permutation, I would shuffle the design matrix and regress MEG data onto the shuffled design matrix. I then threshold and select the largest cluster from it, do this 10,000 or so, and make my distribution. then, I find my p-value based on that distribution

dengemann, 2015-07-10:
What kind of contrast are you talking about?
Are you talking about group data or single subject data?

teonlamont, 2015-07-10:
single subject data. basically I've done a regression on my time-course for all sensors, and I would like to correct the multiple comparisons of the sensors and independent tests per time.

dengemann, 2015-07-10:
Ok I think I got it.
So where do you see a way to test for exchangeability (null hypothesis).
Do you have conditions in your case?

teonlamont, 2015-07-10:
the null hypothesis is that there isn't a correspondence between let's say reaction time and brain activity. if you shuffle your design matrix of reaction time data and you regress it on to MEG data, then you don't expect there to be a relationship between the two. the significance test of the alternative hypothesis is how different is the regression of the observed reaction time on brain activity from the shuffled ones.

dengemann, 2015-07-10:
I think we had this before, the permutation clustering does not apply here
as you cannot formulate an exchangeability hypothesis.
Unless you have some other condition in which the reaction time is affected
or there is no correlation.

teonlamont, 2015-07-10:
what do you mean by an exchangeability hypothesis?

mmagnuski, 2015-07-10:
Isn't the exchangeability hypothesis in this case just the exchangeability of predictor - outcome pairs? If you assume there is no relationship between predictor(s) and outcome they can be permuted to get the null hypotesis distribution.
This is also the case if you used group data but a multilevel regression model (not subject-individual models) - you would have to permute (usually within subjects) the predictor - outcome pairs.
Or maybe I am wrong - and this approach is flawed?

teonlamont, 2015-07-10:
@mmagnuski this is my understanding of it as well. that means this approach shouldn't be problematic for you using this function. it's just at the moment it's not supported, but I wanted to make sure I'm not missing anything.

teonlamont, 2015-11-02:
ok, bringing this discussion up again. @dengemann, you and I briefly talked about this offline. The idea is that if you shuffle your regressors for the permutations, you would expect to your permuted regression models to have low predictive power, i.e. the t-values will be small, p-values will be large. you can use this information to develop your distributions you would then use for the rest of the tests. 

my question is what format would the regression stats function need to be in. should it take in a single time point per sensor/vertex and what should it spit out? it seems that all it needs is a stats function to work nicely with the spatiotemporal cluster test. i'm trying to follow the stats module code but it is a bit complicated :/

teonlamont, 2015-12-13:
after discussing this with a few people, I think the best solution is to have an SPM approach with two levels for analyses: first, we can get use t-values from the regression function as the first level and for perform the spatiotemporal testing at group-level on them. Closing this for now. feel free to reopen

closed 2015-12-13 by teonlamont