mne-tools/mne-python #2273: Error during ica.fit() of raw data

neukym, 2015-07-11:
I am trying to run code from the [ICA example](http://martinos.org/mne/stable/auto_examples/preprocessing/plot_ica_from_raw.html) on my raw data:

```
import sys
sys.path.append('/imaging/local/software/mne_python/v0.9full')
sys.path.append('/imaging/local/software/python_packages/scikit-learn/v0.16.1')
sys.path.append('/imaging/local/software/python_packages/pysurfer/v0.4')
import numpy as np
import mne
from mne.io import Raw
from mne.preprocessing import ICA
from mne.preprocessing import create_ecg_epochs, create_eog_epochs

# Set up data
raw_fname = 'my_raw_file.fif'
raw = Raw(raw_fname, preload=True)
raw.filter(1, 45, n_jobs=2)

# set picks
picks = mne.pick_types(raw.info, meg=True, eeg=False, eog=False, ecg=True, stim=False, exclude='bads')

###############################################################################
# 1) Fit ICA model using the FastICA algorithm
	
ica = ICA(n_components=0.95, method='fastica')
ica.fit(raw, picks=picks, decim=3, reject=dict(mag=4e-12, grad=4000e-13))

```

This code works with the sample "audvis" data, but I get the following error at

`ica.fit(raw, picks=picks, decim=3, reject=dict(mag=4e-12, grad=4000e-13))`

 when I try to run it on my own `_raw.fiff` files:

```
Fitting ICA to data using 308 channels. 
Please be patient, this may take some time
    Rejecting  epoch based on MAG : ['MEG0121', 'MEG1411', 'MEG1421']
Artifact detected in [32000, 34000]
    Rejecting  epoch based on MAG : ['MEG0121', 'MEG1411']
Artifact detected in [62000, 64000]
...
    Rejecting  epoch based on MAG : ['MEG0121', 'MEG1411']
Artifact detected in [920000, 922000]
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-23-7b2d16883d9c> in <module>()
----> 1 ica.fit(raw, picks=picks, decim=1, reject=dict(mag=4e-12, grad=4000e-13), verbose=True)

/imaging/local/software/mne_python/v0.9full/mne/preprocessing/ica.pyc in fit(self, inst, picks, start, stop, decim, reject, flat, tstep, verbose)

/imaging/local/software/mne_python/v0.9full/mne/utils.pyc in verbose(function, *args, **kwargs)
    536         # set it back if we get an exception
    537         try:
--> 538             return function(*args, **kwargs)
    539         finally:
    540             set_log_level(old_level)

/imaging/local/software/mne_python/v0.9full/mne/preprocessing/ica.pyc in fit(self, inst, picks, start, stop, decim, reject, flat, tstep, verbose)
    317         if isinstance(inst, _BaseRaw):
    318             self._fit_raw(inst, picks, start, stop, decim, reject, flat,
--> 319                           tstep, verbose)
    320         elif isinstance(inst, _BaseEpochs):
    321             self._fit_epochs(inst, picks, decim, verbose)

/imaging/local/software/mne_python/v0.9full/mne/preprocessing/ica.pyc in _fit_raw(self, raw, picks, start, stop, decim, reject, flat, tstep, verbose)
    375                                                     raw.info, picks)
    376 
--> 377         self._fit(data, self.max_pca_components, 'raw')
    378 
    379         return self

/imaging/local/software/mne_python/v0.9full/mne/preprocessing/ica.pyc in _fit(self, data, max_pca_components, fit_type)
    459             full_var = np.var(data, axis=1).sum()
    460 
--> 461         data = pca.fit_transform(data.T)
    462 
    463         if isinstance(self.n_components, float):

/imaging/local/software/anaconda/1.9.1/x86_64/lib/python2.7/site-packages/sklearn/decomposition/pca.pyc in fit_transform(self, X, y)
    576 
    577         """
--> 578         X = self._fit(atleast2d_or_csr(X))
    579         X = safe_sparse_dot(X, self.components_.T)
    580         return X

/imaging/local/software/anaconda/1.9.1/x86_64/lib/python2.7/site-packages/sklearn/utils/validation.pyc in atleast2d_or_csr(X, dtype, order, copy, force_all_finite)
    132     """
    133     return _atleast2d_or_sparse(X, dtype, order, copy, sparse.csr_matrix,
--> 134                                 "tocsr", force_all_finite)
    135 
    136 

/imaging/local/software/anaconda/1.9.1/x86_64/lib/python2.7/site-packages/sklearn/utils/validation.pyc in _atleast2d_or_sparse(X, dtype, order, copy, sparse_class, convmethod, force_all_finite)
    109     else:
    110         X = array2d(X, dtype=dtype, order=order, copy=copy,
--> 111                     force_all_finite=force_all_finite)
    112         if force_all_finite:
    113             _assert_all_finite(X)

/imaging/local/software/anaconda/1.9.1/x86_64/lib/python2.7/site-packages/sklearn/utils/validation.pyc in array2d(X, dtype, order, copy, force_all_finite)
     91     X_2d = np.asarray(np.atleast_2d(X), dtype=dtype, order=order)
     92     if force_all_finite:
---> 93         _assert_all_finite(X_2d)
     94     if X is X_2d and copy:
     95         X_2d = safe_copy(X_2d)

/imaging/local/software/anaconda/1.9.1/x86_64/lib/python2.7/site-packages/sklearn/utils/validation.pyc in _assert_all_finite(X)
     25     if (X.dtype.char in np.typecodes['AllFloat'] and not np.isfinite(X.sum())
     26             and not np.isfinite(X).all()):
---> 27         raise ValueError("Array contains NaN or infinity.")
     28 
     29 

ValueError: Array contains NaN or infinity.
```

I can't work out what's causing it. Thanks for any help or suggestions.

dengemann, 2015-07-12:
This is indeed weird and sounds like a low level issue.... I've never had this before. Could you share some of your data? You're getting NaN values in your data at some point ...

neukym, 2015-07-12:
Thanks Denis. The raw file is data is [here](https://sandbox.kymata-atlas.org/myrawfile_raw_sss_movecomp_tr.fif) (temporarily - I will take it down once we know what the problem is). It is maxfiltered MEG/EEG/ECG/VEOG/HEOG Neuromag data (the unmaxfiltered `_raw.fiff` has the same problem - but I can supply this if needed).

Perhaps wierder still is that 5% of the time it *does* work, but it seems non-deterministic as to when this happens. But this might make more sense if there is some sort of randomisation function is called inside ica.fit()?

teonlamont, 2015-07-12:
@neukym you can set a seed in `mne.preprocessing.ICA(random_state=42` if you would like to try to reproduce the same error

neukym, 2015-07-13:
Ah, great, well, that explains the (psuedo) nondeterminism.

neukym, 2015-07-13:
And, just to confirm, that does produce the same error. I guess there must be a set of `random_states` that don't reproduce this error, on this dataset at least.

Eric89GXL, 2015-07-13:
Hmm, I think that points to some bug with the ICA routines. So it works for some random seeds, but not for others? I wonder if we could share some minimal set of data with the people who make the ICA (`sklearn`?) to help them debug it. @dengemann any ideas?

neukym, 2015-07-13:
> So it works for some random seeds, but not for others?

No actually - I just went through testing different seeds (1,2,3.. etc). It worked with seed 5, but then wouldn't work when I re-ran seed 5.

But this might not be a surprise - it depends how the random values are generated from the seeds. Matlab `rand()` seeds are relative to the last called seed (so calling `rand(seed=5)` multiple times gives different random numbers), and it might be the same here.

But I think the underlying intuition - that it works with some initial random numbers, but not others, is correct.

Eric89GXL, 2015-07-13:
Hmm, it's not supposed to work that way. Doing `np.random.RandomState(0).rand(10)` will always give the same 10 numbers. Getting different results with the same seed is indicative of some bug in our code I think, since it means we are not using it correctly.

neukym, 2015-07-21:
Hold up - I am still using sklearn v0.14.1. Will check if this is the problem and report back.

(Our Anaconda build still bundles v0.14.1 of sklearn, and so the line

`sys.path.append('/imaging/local/software/python_packages/scikit-learn/v0.16.1')`

should be

`sys.path.insert(1,'/imaging/local/software/python_packages/scikit-learn/v0.16.1')`)

neukym, 2015-07-23:
OK, with the latest version of Anaconda (and sklearn v0.16.1), I still have the same problem above. A slightly reworded version of the error:
```
Input contains NaN, infinity or a value too large for dtype('float64').
```
on most (70%?) of attempts, and the different but related problem that `random_state=x` doesn't seem to be generating the same random set of initial conditions for the ICA (cycling through seeds, `random_state=6` worked for me just now, but then didn't the second time I ran it).

Someone might want to confirm both of these on their setups though.

agramfort, 2015-07-23:
did you check that you don't have too strong artifacts in your data? They
can cause numerical issues. That's why we have a reject param in ICA object.

neukym, 2015-07-24:
Fair question. I did check and it seems fine: I really don't really have a notion as to what 'too strong artefacts' are, but - 

- The data looks fine (apart from the ECG component, the occasional eyeblink, and a couple of bad channels):
[image]
- I exclude bad channels and EEG from the picks and did use the reject param (`ica.fit(raw, picks=picks, decim=3, reject=dict(mag=4e-12, grad=4000e-13))`) - what is left should be pretty clean. From the logs it doesn't seem to be rejecting any more or less than the example data.
- On those random, handful of times that ICA is computed sucessfully, it picks out the correct ECG components pretty well (as far as I can tell), implying both that the data isn't too laden with artefacts, and enough non-rejected data is getting through.

Please let me know if there is anything more you would like me to check.

agramfort, 2015-07-24:
how did you install your python stack? especially numpy as scipy?

neukym, 2015-07-29:
We have the latest version of [Anaconda](https://store.continuum.io/cshop/anaconda/) on our system (2.3.0), which bundles the latest versions of Scikit-learn, Scipy and Numpy (among other modules):

- Scipy 0.15.1
- Numpy 1.9.2
- Scikit learn 0.16.1

It was installed using the following command set:

```
cd /path/to/anaconda/install/
mkdir ./2.3.0/
cd ./2.3.0/
wget https://3230d63b5fc54e62148e-c95ac804525aac4b6dba79b00b39d1d3.ssl.cf1.rackcdn.com/Anaconda-2.3.0-Linux-x86_64.sh
chmod u+x ./Anaconda-2.3.0-Linux-x86_64.sh
Bash ./Anaconda-2.3.0-Linux-x86_64.sh
```

I was previously using Anaconda 1.9.1, but the errors [seem to be the same in both cases](https://github.com/mne-tools/mne-python/issues/2273#issuecomment-124101680).

Hope this helps. What would be really useful is if someone else could try and reproduce the error on their own system (using the file provided [above](https://github.com/mne-tools/mne-python/issues/2273#issuecomment-120749838)) - if no-one else gets the above error then we'll know the problem must be at my end somehow.

agramfort, 2015-07-30:
looks like the only reason to explain this issue is the file. Could you
share the fif file so we can try to reproduce?

neukym, 2015-07-30:
Sure. The raw file is [here](https://sandbox.kymata-atlas.org/myrawfile_raw_sss_movecomp_tr.fif) (temporarily - I will take it down once we know what the problem is). It is maxfiltered MEG/EEG/ECG/VEOG/HEOG Neuromag data (the unmaxfiltered `_raw.fiff` has the same problem - but I can supply this if needed).

jmontoyam, 2015-08-27:
@neukym, @agramfort 
I can't reproduce this error. I have downloaded the file myrawfile_raw_sss_movecomp_tr.fif, and I have run fastica on it, and the code runs without errors. 
You have mentioned that you are using anaconda. Could you please upload your conda environment info in order to re-create the same env?...this way, I can re-create the same conda env that you are using, and try to reproduce the error. To accomplish this, go to the shell and use the following line:
conda env export > environment.yml
Please upload the resulting environment.yml file.

neukym, 2015-08-28:
Thanks @jmontoyam - *really* appreciated.

Your finding that this works on your system has made us investigate this closer at our end, and we have tracked done some `infs` turning up in `data` in `_fit()`, which may be causing the problem. But we are investigating when and why this occurs. (and why it is only on our data, only with our system, why it is seemingly non-deterministic etc.)

I will update this issue next week with what we find.

The `.yml` is below:

```
name: root
dependencies:
- _license=1.1=py27_0
- abstract-rendering=0.5.1=np19py27_0
- alabaster=0.7.3=py27_0
- anaconda=2.3.0=np19py27_0
- argcomplete=0.8.9=py27_0
- astropy=1.0.3=np19py27_0
- babel=1.3=py27_0
- bcolz=0.9.0=np19py27_0
- beautiful-soup=4.3.2=py27_0
- binstar=0.11.0=py27_0
- bitarray=0.8.1=py27_0
- blaze-core=0.8.0=np19py27_0
- blz=0.6.2=np19py27_1
- bokeh=0.9.0=np19py27_0
- boto=2.38.0=py27_0
- bottleneck=1.0.0=np19py27_0
- cairo=1.12.18=4
- cdecimal=2.3=py27_0
- certifi=14.05.14=py27_0
- cffi=1.1.0=py27_0
- clyent=0.3.4=py27_0
- colorama=0.3.3=py27_0
- conda=3.14.1=py27_0
- conda-build=1.14.1=py27_0
- conda-env=2.2.3=py27_0
- configobj=5.0.6=py27_0
- cryptography=0.9.1=py27_0
- curl=7.43.0=0
- cython=0.22.1=py27_0
- cytoolz=0.7.3=py27_0
- datashape=0.4.5=np19py27_0
- decorator=3.4.2=py27_0
- docutils=0.12=py27_0
- dynd-python=0.6.5=np19py27_0
- enum34=1.0.4=py27_0
- fastcache=1.0.2=py27_0
- flask=0.10.1=py27_1
- fontconfig=2.11.1=4
- freetype=2.5.2=2
- funcsigs=0.4=py27_0
- gevent=1.0.1=py27_0
- gevent-websocket=0.9.3=py27_0
- greenlet=0.4.7=py27_0
- grin=1.2.1=py27_1
- h5py=2.5.0=np19py27_3
- hdf5=1.8.15.1=1
- idna=2.0=py27_0
- ipaddress=1.0.7=py27_0
- ipython=3.2.0=py27_0
- ipython-notebook=3.2.0=py27_0
- ipython-qtconsole=3.2.0=py27_0
- itsdangerous=0.24=py27_0
- jdcal=1.0=py27_0
- jedi=0.8.1=py27_0
- jinja2=2.7.3=py27_1
- jpeg=8d=0
- jsonschema=2.4.0=py27_0
- libdynd=0.6.5=0
- libffi=3.0.13=0
- libpng=1.6.17=0
- libsodium=0.4.5=0
- libtiff=4.0.2=1
- libxml2=2.9.2=0
- libxslt=1.1.28=0
- llvmlite=0.5.0=py27_0
- lxml=3.4.4=py27_0
- markupsafe=0.23=py27_0
- matplotlib=1.4.3=np19py27_2
- mistune=0.5.1=py27_1
- mock=1.0.1=py27_0
- multipledispatch=0.4.7=py27_0
- networkx=1.9.1=py27_0
- nltk=3.0.3=np19py27_0
- nose=1.3.7=py27_0
- numba=0.19.1=np19py27_0
- numexpr=2.4.3=np19py27_0
- numpy=1.9.2=py27_0
- odo=0.3.2=np19py27_0
- openpyxl=1.8.5=py27_0
- openssl=1.0.1k=1
- pandas=0.16.2=np19py27_0
- patsy=0.3.0=np19py27_0
- pep8=1.6.2=py27_0
- pillow=2.8.2=py27_0
- pip=7.0.3=py27_0
- pixman=0.26.2=0
- ply=3.6=py27_0
- psutil=2.2.1=py27_0
- ptyprocess=0.4=py27_0
- py=1.4.27=py27_0
- py2cairo=1.10.0=py27_2
- pyasn1=0.1.7=py27_0
- pycosat=0.6.1=py27_0
- pycparser=2.14=py27_0
- pycrypto=2.6.1=py27_0
- pycurl=7.19.5.1=py27_2
- pyflakes=0.9.2=py27_0
- pygments=2.0.2=py27_0
- pyopenssl=0.15.1=py27_1
- pyparsing=2.0.3=py27_0
- pyqt=4.11.3=py27_1
- pytables=3.2.0=np19py27_0
- pytest=2.7.1=py27_0
- python=2.7.10=0
- python-dateutil=2.4.2=py27_0
- pytz=2015.4=py27_0
- pyyaml=3.11=py27_1
- pyzmq=14.7.0=py27_0
- qt=4.8.6=3
- readline=6.2=2
- redis=2.6.9=0
- redis-py=2.10.3=py27_0
- requests=2.7.0=py27_0
- rope=0.9.4=py27_1
- runipy=0.1.3=py27_0
- scikit-image=0.11.3=np19py27_0
- scikit-learn=0.16.1=np19py27_0
- scipy=0.15.1=np19py27_0
- setuptools=17.1.1=py27_0
- sip=4.16.5=py27_0
- six=1.9.0=py27_0
- snowballstemmer=1.2.0=py27_0
- sockjs-tornado=1.0.1=py27_0
- sphinx=1.3.1=py27_0
- sphinx_rtd_theme=0.1.7=py27_0
- spyder=2.3.5.2=py27_0
- spyder-app=2.3.5.2=py27_0
- sqlalchemy=1.0.5=py27_0
- sqlite=3.8.4.1=1
- ssl_match_hostname=3.4.0.2=py27_0
- statsmodels=0.6.1=np19py27_0
- sympy=0.7.6=py27_0
- system=5.8=2
- terminado=0.5=py27_0
- theano=0.7.0=np19py27_0
- tk=8.5.18=0
- toolz=0.7.2=py27_0
- tornado=4.2=py27_0
- ujson=1.33=py27_0
- unicodecsv=0.9.4=py27_0
- util-linux=2.21=0
- werkzeug=0.10.4=py27_0
- xlrd=0.9.3=py27_0
- xlsxwriter=0.7.3=py27_0
- xlwt=1.0.0=py27_0
- yaml=0.1.6=0
- zeromq=4.0.5=0
- zlib=1.2.8=0
- pip:
  - backports.ssl-match-hostname==3.4.0.2
  - beautifulsoup4==4.3.2
  - blaze==0.8.0
  - sphinx-rtd-theme==0.1.7
  - tables==3.2.0

```

jmontoyam, 2015-08-31:
@neukym, @agramfort
I have re-created the conda environment according to the environment.yml mentioned before.
By taking a close look at the environment.yml file I have realized that this environment does not contain the mne_python package. 
@neukym, are you sure this is the environment that you use when you run fastica on your data?

Anyway, I have tried to reproduce the issue: I have re-created the same environment mentioned in the 
environment.yml file, and I have installed the mne_python package (0.10.dev0). Then, I run this tiny test on the raw file provided by @neukym:

-------------------------------------------------------------------------------------------------------------------------------------------
import mne
from mne.io import Raw
from mne.preprocessing import ICA

raw_fname = 'myrawfile_raw_sss_movecomp_tr.fif'
raw = Raw(raw_fname, preload=True)
raw.filter(1, 45, n_jobs=2)

ica = ICA(n_components=0.95, method='fastica')
picks = mne.pick_types(raw.info, meg=True, eeg=False, eog=False, stim=False, exclude='bads')
ica.fit(raw, picks=picks, decim=3, reject=dict(mag=4e-12, grad=4000e-13))

--------------------------------------------------------------------------------------------------------------------------------------------

This tiny test runs without errors.

@neukym, please verify that the environment.yml file that you have uploaded belongs to the same conda environment that you use when you try to run fastica on your data. The environment.yml file should contain a line with the mne_python package info.

Just in case, please run again the previously mentioned tiny test, and upload again the environment.yml (please be sure this file contains the mne_python package info).

neukym, 2015-08-31:
> By taking a close look at the environment.yml file I have realized that this environment does not contain the mne_python package. 

You are right @jmontoyam, I normally import it using `sys` at the beginning of the script:

```
import sys
sys.path.append('/imaging/local/software/mne_python/v0.9full')
```

You have made me think about this copy of mne_python on our system - someone other than me installed this package, and I think that, despite the name, it may not be a stable version of v9, but an unstable version of v10! 

I will run your test using my own local copy of 0.10.dev0 (adding it to the conda environment, instead of in the script).

neukym, 2015-09-03:
Hi @jmontoyam - not only does your test work, but it explains why I was getting problems: I was erroneously including `ecg=True` in my `picks`, when this should be left out. Should have picked that up earlier! Everything seems to work fine now.

Thanks for your help - much appreciated :-)

closed 2015-09-03 by agramfort