mne-tools/mne-python #2275: Time Frequency functionality

teonlamont, 2015-07-12:
I just started to use the time frequency module and I have noticed some things that can be improved. two big things, plotting functions should be centralized, and time frequency functions should all take picks as an argument.

- [ ] plotting functions should be in moved to viz
- [ ] probably refactored with overlapping plotting functions
- [ ] picks should be added to all functions. this would be useful for computation on one channel type. for me, it would be useful when plotting the time frequency of IC sources.

teonlamont, 2015-07-12:
one more thing is that I feel like picks should be the first non-obligatory argument in the function. that would mean four api function changes. I could just put it at the end but I hate badly organized arguments :/

Eric89GXL, 2016-03-16:
I'm moving this to 0.13 unless someone wants to volunteer for it

Eric89GXL, 2016-09-08:
@teonbrooks any time for this in the next two weeks?