mne-tools/mne-python #2277: Maxwell Filter

wronk, 2015-07-13:
Next round of improvements for Maxwell filtering:

1) Check that `raw` files play nice with covariance calculations after Maxwell filtering.
2) Add a check that `head_dev_t` (head to device transformation matrix) is defined within the `raw.info`
3) Exclude bad coils from multipolar moment estimation. Reconstruct data at bad coils.
4) Refactor forward code used so code isn't repeated (`maxwell._make_coils` function)
5) Track down all the bookkeeping Elekta's Maxfilter adds and update `maxwell._update_info`
6) Add fine calibration code (3 subparts) with help from Samu
7) Discuss what to do about coil definitions for other MEG systems

wronk, 2015-07-13:
@teonlamont @Eric89GXL @dengemann, I think I've added all the issues you've collectively mentioned, but have a quick look to make sure.

Eric89GXL, 2015-07-13:
Add one at the top preventing use in systems with compensation until we
figure out how to do it properly. Or @staulu @agramfort do you think it
should be easy to get that bit working?

Eric89GXL, 2015-07-13:
@wronk I edited your comment to make it look a little bit nicer, using `1.` instead of `1)` makes use of the markdown support

agramfort, 2015-07-13:
I know too little about compensation to have a good answer...

staulu, 2015-07-13:
A quick question for clarification before I give further comments: Does compensation refer to movement compensation or Elekta's internal active compensation ("MaxShield"), or something else? Another question: How about implementing tSSS? Many people find it the most useful feature of MaxFilter.

Eric89GXL, 2015-07-13:
+1 for tSSS, it would indeed be great

When I say "compensation" I mean the compensation used by CTF systems. I don't know too much about it, but I know we need to be careful e.g. in forward modeling about how to deal with it, so I figured something similar might apply here.

Eric89GXL, 2015-07-13:
i.e., this:

http://martinos.org/mne/stable/manual/convert.html#applying-software-gradient-compensation

wronk, 2015-07-13:
tSSS added :)

I don't have much of an intuition for the compensation either... How high of a priority is it include support for this?

Eric89GXL, 2015-07-13:
I think the first thing you should do is add support for raising an error if someone tries to use it. Actually getting it to work... I'm not sure. Lower than that :) I think it should go below most of the other things (calibration / cross-talk, tSSS) because we need those features before this is really useable for people.

Eric89GXL, 2015-07-13:
List reordered as a potential roadmap, @wronk @staulu @agramfort @dengemann see if it makes sense to you

wronk, 2015-07-13:
LGTM

wronk, 2015-07-13:
As for the 6th point (coil definitions), I think that might be a good first test for the localization evaluation code @yousrabk and I are working on this week. Does anyone know where we might get access to coil any/all of the geometry drawings/blueprints? 

Besides for the main issue of selecting integration points, there is a small discrepancy between Elekta and MNE with the size of the gradiometer baselines. I don't think both can be correct, and @staulu mentioned he based his numbers off some gradiometer blueprints when working with Elekta.

Eric89GXL, 2015-07-13:
@mshamalainen where does the baseline for the gradiometers in the `coil_def.dat` file come from? There seems to be a small discrepancy between the MNE values and those from Elekta.

@wronk actually I'm not sure it's worth pursuing coil defs for other systems for now. I'll move that point to later. I think it's worth investigating for Elekta because we need it for our sanity, but let's wait until we see gains / big differences before bothering to add defs for other systems.

staulu, 2015-07-13:
The list of tasks looks good to me. As for the reference sensor compensation, such as the 3rd order gradiometrization in CTF systems, the compatibility of the compensated data with MaxFilter depends on how the compensation is done. As far as I understand, the CTF compensation is done by suppressing the homogeneous fields and the 1st and 2nd order derivatives of the external fields based on the reference sensor recordings. This kind of data can be processed with MaxFilter without any extra manipulations and the resulting data should have any required forward-model reconstructions automatically applied, just like in the case of Elekta's active compensation.

Eric89GXL, 2015-07-13:
So if I understand correctly, we should be able to process compensation coils just as we process any other coils, and things should proceed normally. At first it seemed strange to me, since the reference sensors are designed to capture external interference, and we will be using Maxwell filtering to suppress those signals. But I suppose it doesn't matter whether the reference sensors or Maxwell filtering does the external artifact suppression, so long as it happens correctly somewhere along the processing chain (and later compensation application does not do something bad when signals have been reduced to near zero, which they shouldn't).

Eric89GXL, 2015-07-13:
Okay, I updated PR1 to say that we should warn that processing non-Elekta data is experimental. This will cover us on systems with reference sensors / compensation implemented. At some point we need to validate that it works for those systems, which I have toward the bottom of the list (low priority for now).

staulu, 2015-07-13:
Your comment made me realize that it is actually not feasible to perform the reference sensor based compensation if SSS has already been applied to the whole CTF array (actual sensors + reference sensors) because all the reconstructed signals will be of internal origin and the assumptions that justify using external Taylor's expansion for the extrapolation of reference signals don't hold true anymore. The solution would be 1) just to apply SSS on the whole array (it will be numerically more stable thanks to the reference sensors) and consider this the final interference suppression step without additional reference sensor compensation, or 2) first perform the 3rd order gradiometrization and then possibly do SSS depending on the details of the compensation. I think this topic requires some extensive thinking and planning, maybe in a future project :)

Eric89GXL, 2015-07-13:
Okay, we can postpone that until we have fine calibration / cross talk,
tSSS, movement compensation, and multipolar moment forward calculation
working :)

teonlamont, 2015-07-14:
btw, the equations are not being rendered in the compensation section. 

http://martinos.org/mne/stable/manual/convert.html#applying-software-gradient-compensation

agramfort, 2015-07-14:
yeah I have seen... all formulas don't render...

wronk, 2015-07-30:
For everyone's info, the plan is to focus the next PR on fine-calibration. Getting that should lead to a 10x improvement in shielding factor and actually make this work useful in real processing pipelines.

Eric89GXL, 2015-07-30:
Great. How should we handle the fine-calibration files that must be supplied? It would be good if we had a solution that allowed us *not* to have to supply it, if it can somehow be autodetected. Can we for example autodetect that a file was recorded at MGH or UW, and use the appropriate file if `cal_file=None` (the default)? Not sure if that's feasible, but it would be awesome if we could. We could fairly easily have the package download the appropriate file once and save it somewhere.

wronk, 2015-07-30:
If we knew the site, then presumably this should be possible. The file usually is named something like `sss_cal.dat` which is a sym link to the actual file like `sss_cal_3053.dat` where the number is a unique ID for each MEG site. I'm not sure if the folder tree containing that file is standardized though... I think a decent solution would be to just have those `dat` files for all the sites in MNE python.

*Eventually*, it would be good to also have the algorithm that generates these files. Apparently, Samu's original method just uses a couple minutes of ERM data to construct all the fine-calibration information.

Eric89GXL, 2015-07-30:
Yeah, that's my idea, we should collect and keep the `.dat` files somewhere, probably auto-downloaded the first time they are needed to keep package size down. The question is, can we automatically determine which site/`ss_cal_*.dat` file each FIF file needs?

dgwakeman, 2015-07-30:
IMO, we don't want to have a set file per site. This file does change with system maintenance etc.. A new one should be generated each time a maintenance occurs. While the changes, I have observed haven't been massive, they do and could occur. Adding additional checks to go through the dates of recordings etc. is messy. 
+1 for generating our own (collecting ER data is easy with each recording).

Eric89GXL, 2015-07-30:
@dgwakeman I'm just thinking that the way it works right now, is there is basically one file per site at any given point in time. If I'm at MGH and I do `maxfilter ...` then I automatically use whatever the latest cal file is. If I'm at UW and I do it, I get a different one, etc. So in some sense, the proposal is to preserve the existing behavior, and in some ways make it better because I won't need to be at the site to get that functionality. You can always select a different file for example if you know your data are old, you can use the old cal file. This seems the least obtrusive method to me.

Generating our own would be okay, but it does require something at acquisition (recording a specific type of data), analysis (providing an ERM file whenever doing SSS), and at the algorithm end (putting together another algorithm to do the computations). This seems harder to me, with minimal gain for non-trivial amount of extra difficulty/risk in errors, at least for a first pass.

Eric89GXL, 2015-07-30:
@staulu how careful should we be about different calibration versions? Is it safe (enough) to assume, as MaxFilter does currently, that we can get away with one `sss_cal.dat` file per site for all recordings from there?

dgwakeman, 2015-07-30:
I guess it is no worse, than the current setup, assuming we have someone at each site updating this file as necessary to us at github. Maybe it would be better if at first run it asked for the location of `sss_cal.dat`, and made a symlink to  and used it by default each time. That way we don't have to check that we have the file for each site up to date here?

wronk, 2015-07-30:
@Eric89GXL, there might be a way to determine the site, but I doubt we can rely on it as data gets anonymized, manipulated, etc.

@dgwakeman, the code for generating our own will be nice but it will probably come in as a feature later down the road. It may also come with more legal questions since that was developed at Elekta...

Symlink would be good. It's usually on the MEG computer system though (requiring an SSH), so it might be difficult, right? It sounds like we build up a few triage levels and fall back on the original (and perhaps out of date) `.dat` file.

Eric89GXL, 2015-07-30:
We'd probably need each user to get the file from their system, which wouldn't be the end of the world. We can store common ones on the Wiki if we want.

Eric89GXL, 2015-07-30:
i.e., you ssh / scp once to get the file on your system, then mne-python stashes it somewhere for future use. It would still be good to have some sanity checking if we could so that we try to avoid mis-application of systems. I think the acquisition machine might be stored in the `info` structure somewhere, in which case the user could enter that name along with the cal file, and only allow auto-use of the cal file if the acquisition machine name matches.

dgwakeman, 2015-07-30:
@wronk, it should be accessible to any machine that can use maxfilter, so it should be on the network somewhere, usually `$NEUROMAG_ROOT/databases/sss/`.

staulu, 2015-07-30:
I am afraid that there is no way to reliably check the site that the sss_cal.dat file corresponds to. We just need to trust that Elekta has shipped the correct file and that nobody has accidentally changed it afterwards. The calibration information should be pretty stable over time, but major hw operations, such as changing of many sensors, could require a new calibration. The most reliable way to check the validity of the calibration is to run SSS on the data and make sure that the environmental interference decreases by a factor of 100 or more.

wronk, 2015-08-11:
@akclee, this github issue outlines all the work that's either complete or left to be done for my GSoC project

Eric89GXL, 2015-12-21:
Closing this since we can pick up the rest in discrete issues/PRs. Thanks @wronk!

closed 2015-12-21 by Eric89GXL