mne-tools/mne-python #2280: more datasets for mne-python

mainakjas, 2015-07-13:
I am opening a new issue to avoid cluttering gitter. Here's what I could find till now. Edit this comment to add more datasets

* [Fieldtrip datasets](http://www.fieldtriptoolbox.org/faq/what_types_of_datasets_and_their_respective_analyses_are_used_on_fieldtrip). They have a couple of examples, we could try to reproduce. For example: [beamforming](http://www.fieldtriptoolbox.org/tutorial/beamformingextended), [artifact rejection](http://www.fieldtriptoolbox.org/tutorial/automatic_artifact_rejection). So, we could get the corresponding datasets
* [UCSD EEG datasets](http://headit-beta.ucsd.edu/studies)

agramfort, 2015-07-13:
artifact rejection one seems doable. Dics uses mat files.

how about brainstorm?

mainakjas, 2015-07-13:
For brainstorm, you need to register and then you can download. But they have dropbox links as well for all the files. Interestingly, they have median nerve data (tutorial [here](http://neuroimage.usc.edu/brainstorm/DatasetMedianNerveYokogawa) and [here](http://neuroimage.usc.edu/brainstorm/Tutorials/TutMindNeuromag)) -- might be cool to demonstrate our stim artifact interpolation, resting state data (tutorial [here](http://neuroimage.usc.edu/brainstorm/Tutorials/Resting)). Also auditory data but maybe that's not so interesting.

Brainstorm has GPL license and so has Fieldtrip. How do we deal with that? Maybe you need to talk to your friends there ;-)

agramfort, 2015-07-14:
+1 for mind somato data

for resting we need more stuff in mne to make it useful.

regarding license there should be no pb.

can we download them from a script?

dengemann, 2015-07-14:
> Dics uses mat files.

alexandrebarachant, 2015-07-16:
There is also the [BNCI data repository](http://bnci-horizon-2020.eu/database/data-sets)
It's mainly EEG / BCI but there is a couple of interesting one (including a fNIRS dataset).
File are all in matlab format, but they are more or less standardized.

agramfort, 2015-07-16:
cool !

what type of cool example would you do with it?

alexandrebarachant, 2015-07-16:
It is probably a good source to feed the decoding section.
the fNIRS data would be interesting to look at. There is also one with joint recording of EEG and EMG data. But i don't have any cool stuff to do with it (yet).

Ultimately, having a generic wrapper for all these dataset will allow easy multi-dataset evaluation of algorithms.

agramfort, 2015-07-16:
> It is probably a good source to feed the decoding section.
> the fNIRS data would be interesting to look at. There is also one with
> joint recording of EEG and EMG data. But i don't have any cool stuff [...]
I like that !

mainakjas, 2015-07-16:
yeah, one related issue that I don't like having to do is:

```py
data_path = sample.data_path()
raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
```
Instead if we could save it as a dict and do:
```py
raw_fname = datapath['raw_fname']
```
that would be nice. Just an idea ...

dengemann, 2015-09-03:
isn't this done now?

Eric89GXL, 2015-09-16:
@dengemann we have brainstorm, but not FieldTrip or UCSD, I think

@mainakjas is it a realistic target to get these done in the next two weeks?

mainakjas, 2015-09-16:
hmmm ... two weeks might not be enough. I can take a look in the Dec holidays if there is no rush. We can shift the milestone.

Eric89GXL, 2015-09-16:
done

Akshay0724, 2017-03-18:
I would like to work in this issue for GSOC, please tell me how do I start working and what exactly needs to be done here.

agramfort, 2017-03-19:
@Akshay0724 see how data downloaders are setup in mne.datasets and then you need to write a tutorial. Basically take any tutorial that you want in:

http://www.fieldtriptoolbox.org/faq/what_types_of_datasets_and_their_respective_analyses_are_used_on_fieldtrip

make the downloader and make an example / tutorial to see how to analyze this data.

Akshay0724, 2017-03-26:
> datasets_and_their_respective_analyses_are_used_on_fieldtrip

Hello @agramfort <https://github.com/agramfort>,
For loading FieldTrip data in our data structures, we can either first
convert data into format compatible with mne.io.read_raw_fif and then make
the converted data available for download
or
we make downloader to get data from FieldTrip and make it responsible for
the conversion.

Please correct me if I'm wrong.
Thanks

agramfort, 2017-03-26:
find a dataset that is in native format eg. .ds, .edf or .fif

choldgraf, 2017-04-07:
another idea is to add in an API to connect with the CRCNS database. It's super simple, see https://github.com/neuromusic/crcnsget

jasmainak, 2017-04-07:
What sort of example do you have in mind @choldgraf ? For the simple use case of getting more datasets, you can use [datalad](http://datalad.org/)

sappelhoff, 2019-08-21:
@jasmainak is this dead? Or related to the generalized openneuro fetcher?

jasmainak, 2019-08-21:
I am not sure. It was related to my mission doctoral and I implemented the Brainstorm fetchers already. There are many other indexes of datasets online, so I'm fine closing this.

agramfort, 2019-08-22:
let's keep this open as a source of new datasets and analysis examples. But yes we should progressively move towards BIDS formatted data.

larsoner, 2023-10-01:
I'm going to close since I don't think we need to implement anything at the MNE end for this at the moment. Let's open issues about specific datasets that people are motivated to add / think MNE needs nowadays

closed 2023-10-01 by larsoner