mne-tools/mne-python #2283: BUG: regression random state ICA

dengemann, 2015-07-13:
In random_state == None will not any longer produce a fixed seed. I suspect some underlying changes in sklearn. @agramfort does this ring a bell?

agramfort, 2015-07-13:
no bell

dengemann, 2015-09-08:
turns out that our defaults == None will not produce a fixed seed. ...

closed 2015-09-09 by agramfort

dengemann, 2015-09-09:
@agramfort we should maybe change that. This is confusing for users. Maybe just fixt the seed or print it.

agramfort, 2015-09-09:
I would not fix it. Users should realize that ICA is a non-deterministic
algo.
I'd just improve the doc

dengemann, 2015-09-09:
Yes, agreed. I think the confusion was arising from the fact that we indeed once had a bug but then fixed it. The examples then went fine as we passed a seed, not because of the defaults. I think I forgot about that.