mne-tools/mne-python #2286: ENH: allow for spatial connectivity between two hemispheres

dengemann, 2015-07-14:
In volumetric source spaces you can get bi-lateral ROIs, blobs, and  connected clusters, not that easily with Freesurfer surfaces. However, it's an unphysiological prior to not allow bihemispheric connections in permutation clustering tests. With @banilo and @yuval-harpaz we were wondering about ways to connect medial vertices across hemispheres, maybe based on a minimum distance criterion. As an intermediate step it would be cool to support FS xhemi registrations to support mapping vertices between hemispheres. Input and ideas are welcome.

cc @Eric89GXL @mainakjas @agramfort

yuval-harpaz, 2015-07-14:
Hi
I used fsaverage_sym from:
https://surfer.nmr.mgh.harvard.edu/fswiki/Xhemi
I read the spheres surfaces with matlab and looked for min distance there,
which made more sense than on pial.
I then looked for each of these pairs of dots (in lPair and rPair vectors
below) for their closes points on individual sphere surfaces.
The result seems reasonable but I did not validate it yet.

This is my code for looking for LR pairs on the template

sphere.lh=mne_read_surface('/usr/local/freesurfer/subjects/fsaverage_sym/surf/lh.sphere.left_right');
sphere.rh=mne_read_surface('/usr/local/freesurfer/subjects/fsaverage_sym/surf/rh.sphere.left_right');
for srci=1:10242

[~,lPair(srci)]=min(sum(abs(sphere.lh-repmat(sphere.rh(srci,:),length(sphere.lh),1)),2));
    prog(srci)
end
rPair=1:10242;

here is how I looked for pairs for individuals

       sub
*Sphere.lh=mne_read_surface(['/usr/local/freesurfer/subjects/',subj,'/surf/lh.sphere']);*
*
subSphere.rh=mne_read_surface(['/usr/local/freesurfer/subjects/',subj,'/surf/rh.sphere']);*
*        clear li ri*
*
%[v,i]=min(sum(abs(sphere.rh-repmat(subSphere.rh(1,:),length(sphere.lh),1)),2));*
*        for srci=1:10242*
*
[~,ri(srci)]=min(sum(abs(subSphere.rh-repmat(sphere.rh(srci,:),length(subSphere.rh),1)),2));*
*
[~,li(srci)]=min(sum(abs(subSphere.lh-repmat(sphere.lh(lPair(srci),:),length(subSphere.lh),1)),2));*
*            prog(srci)*
*        end*
*        disp(' ');*
*        if length(unique(li))<10242*
*            disp([subj,' has ',num2str(10242-length(unique(li))),'
repeting matches for LH'])*
*        end*
*        if length(unique(ri))<10242*
*            disp([subj,' has ',num2str(10242-length(unique(ri))),'
repeting matches for RH'])*
*        end*

Eric89GXL, 2015-07-14:
A while back I used Euclidean distance for the medial wall labels, it's
pretty easy to combine that with the existing surface distances. Just
compute the distances between medial left vertices and medial right
vertices, and populate the correct entries in the existing distance matrix.

Eric89GXL, 2015-11-25:
I think this can be done in a few lines with e.g. `cdist`. Okay to close this one as an mne-python issue, or do you think the API should live in mne-python?

dengemann, 2015-11-25:
> do you think the API should live in mne-python?

it should once we're confident that his does the job.

Eric89GXL, 2015-11-25:
Can you come up with the desired API call? `get_euclidean_inter_hemi_neighbors(src, limit)` or so? Then your connectivity matrix would just be the surface one union the euclidean one.

dengemann, 2015-11-25:
Maybe ```mne.connect_hemis(connectivity, src, limit)```

yuval-harpaz, 2015-11-25:
by the way, I played with corregistering left to right hemispheres some
more a while ago. I found that the simplest way is to corregister one
hemisphere of a subject to the other, and then on the inflated overlaping
surfaces (say the real LH and the flipped RH) to find nearest nodes

PR #2643 (2015-12-07): MRG: Inter-hemisphere Euclidean distance connectivity
@dengemann can you see if it makes sense for your use case? I went with naming that was more consistent with the existing spatial distance functions than what we talked about. But it should be pretty straightforward to combine connectivity matrices.

Closes #2286.
closed 2015-12-07 by dengemann