mne-tools/mne-python #2287: Document info data structure

wmvanvliet, 2015-07-14:
For being such an important data structure, there sure isn't any documentation about the `Info` class. What fields might it contain? What do they mean? I know the answer often is 'it depends', but the standard fields should be documented.

dengemann, 2015-07-14:
@wmvanvliet calls for some tutorial or so ... you volunteer, right?

wmvanvliet, 2015-07-14:
yes, I'm working on it right now.

Eric89GXL, 2015-07-14:
I just had to explain some of these the other day. I'll be on gitter soon
if you have questions about the fields.

dgwakeman, 2015-07-14:
Also, this is very helpful:
http://martinos.org/mne/stable/manual/matlab.html?highlight=meas%20info%20structure
Matti's documentation is always super thorough

wmvanvliet, 2015-07-14:
@dgwakeman  that is **really** helpful!

Eric89GXL, 2015-07-15:
cc @mdclarke there are some links in here about the `Info` structure, but @wmvanvliet is planning to add more docs soon. @wmvanvliet are you working on it as part of the sprint?

wmvanvliet, 2015-07-16:
yes, you can find our progress here:
https://github.com/wmvanvliet/mne-doc-notebooks

mainakjas, 2015-07-16:
Great @wmvanilet for giving this a shot! Some feedback:

- Have you looked at the mne-tutorials that already exist?
- I wouldn't start the data structures tutorial with the info. I would
rather say raw -> epochs -> evoked and then describe info.
- Is this a tutorial or a manual thing? If it's a manual thing, you have to
be more comprehensive and it shouldn't have so much code / code output. You
don't need to have code that is "executable" in a manual. It can be just
snippets (at least in my mind). For example, you should say what *each*
field stands for even if it may sound obvious from the name
- "Creating data structures from *scratch* " sounds a bit misleading. The
user may try to go there first thinking it's the first tutorial they should
do, but clearly it's not.

... more comments to follow later.

PR #2642 (2015-11-25): MRG+1: Fix information about info
Just filling out some last bits of information about `Info`.

Closes #2287.
closed 2015-11-25 by dengemann