mne-tools/mne-python #2308: Scale MRI is broken

teonlamont, 2015-07-16:
Scaling subject is currently not working. At first, I thought it was related to my pr #2239 but it isn't working on master either. This may be a problem with the gui interaction with qt (at least this error message shows up on qt listservs).

```
2015-07-16 15:03:43.080 python[21563:3737877] modalSession has been exited prematurely - check for a reentrant call to endModalSession:
```

Pseudo-way to make it happen:
mne.gui.coregistration()
scale a subject without MRI to template using their headshape info
try to save

cc: @christianbrodbeck

christianbrodbeck, 2015-07-19:
Is it working with mne 0.8/0.9? Or is it a problem of another library's update?

Eric89GXL, 2015-10-19:
I'm going to move this to 0.11, let me know if you have time for it in the next couple of days @teonlamont and we can move it back

Eric89GXL, 2016-04-20:
@teonbrooks any updates on this? It has been open for a while with no updates.

Eric89GXL, 2016-09-08:
Closing due to non-response, @teonbrooks feel free to reopen if you hit the issue again

closed 2016-09-08 by Eric89GXL