mne-tools/mne-python #2310: Epochs constructor (re-creates) and re-applies proj even if proj is False if raw.proj is True

haribharadwaj, 2015-07-17:
This leads to problems in at least one use case where the user is interested in working with only a subset of channels. For example:

```
raw.apply_proj()  # Apply proj before selecting channels of interest

raw2 = raw.pick_channels([...], copy=True)   # Pick subset of channels
# Now raw2.proj is True

epochs = Epochs(raw2, ..., proj=False)
# The data in epochs is messed up owing to re-application
# of inappropriate projector without having all channels
```

There are several places here where at least warnings should have been thrown...

haribharadwaj, 2015-07-17:
I think the issue is more general.. projs should be discarded when picks is used at anytime as the old proj definitions wont be valid anymore.. 

What am I missing?

haribharadwaj, 2015-07-17:
Example:
```
# Let's say raw.proj is False
epochs = Epochs(raw, ..., proj=True, picks=[...])  # Pick subset of channels
```
Here picks happens before projections are activiated and data is messed up as result... Thus picks is not useable at all with projs!!

agramfort, 2015-07-17:
oh boy indeed...

2 options I see:

1/ we prevent channel dropping/adding if some proj are active
2/ we automatically reapply active projs

what do you think?

agramfort, 2015-07-17:
suggestion. We add a parameter to drop_channels et al. methods. For
example ignore_proj. If ignore_proj=False (default) and inst.proj ==
True then throw an error. If the "User" sets ignore_proj=True then he
can bypass the check

wdyt?

teonlamont, 2015-07-17:
kinda related question, why is it that we cannot remove the projection once it's been applied? wouldn't it just be multiplying the inverse?

Eric89GXL, 2015-07-17:
The projection matrix is non-invertible. You're basically creating a hole in the data, once it's there, there is no way to know what used to be there. Think about zeroing out a single time point (i.e., the [1, 0, 0, ..., 0] time basis vector) across all channels instead of a single spatial component across all times, and it will be clearer. Once you zero it out, you can't know what you need to put back.

teonlamont, 2015-07-17:
ah, I forgot about the invertibility consideration. makes sense. thanks @Eric89GXL!

haribharadwaj, 2015-07-17:
@teonlamont : Projection is like taking a pont in 3D space and mapping onto it's corresponding 2D (like, (2,3, 5) --> (2, 3)).. No way to recover lost info..

haribharadwaj, 2015-07-17:
Anyways... I agree that picks/drops functions should by default ignore proj AND also warn user AND also rewrite the resulting instance's proj attribute if possible so that other functions don't reapply them.. Seems like edits maybe needed in many places?

agramfort, 2015-07-19:
PR highly appreciated @*haribharadwaj <https://github.com/haribharadwaj> *

Eric89GXL, 2015-12-21:
I think currently we actually support applying projections if channels are missing -- see what `mne_browse_raw` does with MEG channels with the acquisiton projectors active, for example. Another thing to keep in mind is that this issue actually goes a bit beyond just picking and dropping, since bad channels are also removed from projection operations, and thus can modify what projection actually gets done. But maybe we shouldn't worry about that case for now.

> We add a parameter to drop_channels et al. methods. For example ignore_proj. If ignore_proj=False (default) and inst.proj ==True then throw an error. If the "User" sets ignore_proj=True then he can bypass the check

> I agree that picks/drops functions should by default ignore proj AND also warn user AND also rewrite the resulting instance's proj attribute (and delete projection definitions from resulting inst) so that other functions don't reapply them

For @haribharadwaj's original use case:
```
raw.apply_proj()  # Apply proj before selecting channels of interest
raw2 = raw.pick_channels([...], copy=True)   # Pick subset of channels
epochs = Epochs(raw2, ..., proj=False)
```
Just to be clear, the proposal is to have `pick_channels(..., ignore_proj=False)` by default, so doing `raw.pick_channels(...)` as above would throw an error, because `raw` has some projectors applied that have missing channels, right?

If the user did `raw2 = raw.pick_channels(..., ignore_proj=True)`, does that mean that the offending projectors will be set to `False`, or removed from the instance altogether?

This is a bit related to #2734, but should probably be addressed in a separate follow-up PR.

agramfort, 2015-12-21:
if we remove projections then they cannot be taken into account in
inverse to be applied to the lead field matrix. this is bad.

if we keep them then they could be reapplied on a subset of channels
which is also bad.

suggestions:
- once a projection is applied, we need to know to what channels it
was applied. So let's remove from the projection the channels ignored
when the proj was applied. At least we have provenance.
- we prevent reapplying a projection if the channels have changed.

do I miss something?  do we still need another parameter?

dengemann, 2015-12-22:
> do I miss something?  do we still need another parameter?

HBM 2015: "Just don't use PCA" ;)

jasmainak, 2015-12-22:
I'm going to say something very annoying here (yet again) ;) We need to document this. The docstrings are good, but it needs to be all in one place. It will also make the thought process clearer. Projections were the most confusing thing for me when I started on MNE. I am making an effort here: https://github.com/mne-tools/mne-python/pull/2728/files but with projections, I am clearly not the best person to describe it.

Eric89GXL, 2015-12-22:
> we prevent reapplying a projection if the channels have changed.

I'm not sure practically how we would achieve this, though, when using e.g. `pick_channels` or `pick_types`.

Eric89GXL, 2016-04-20:
I can't think of a great solution for this right now. Currently we at least raise a warning about projection being dangerous. I'm going to move it to 0.13 unless someone has a good proposal

Eric89GXL, 2016-09-08:
@agramfort the last suggestion was:

> - once a projection is applied, we need to know to what channels it was applied. So let's remove from the projection the channels ignored when the proj was applied. At least we have provenance.
> - we prevent reapplying a projection if the channels have changed.

I think we need to add to this:

- raise an error in places in the code (probably anything that uses whitening, like inversion and dipole fitting?) where projection information is necessary
- don't set `proj = True` in `Epochs` constructor based on `raw.proj` anymore

Does that make sense? If so I can try to tackle this for 0.13.

agramfort, 2016-09-10:
I 'think' it makes sense. But I would see by trying what are the consequences...

Eric89GXL, 2017-02-14:
See also #3664