mne-tools/mne-python #2311: Raw indexing error

wronk, 2015-07-17:
`raw.time_as_index` function can produce incorrect indices because it truncates instead of rounds when finding the nearest time value.

```
(Pdb) raw
<RawFIF  |  n_channels x n_times : 376 x 14400>
(Pdb) ts = raw.times
(Pdb) ts
array([  0.00000000e+00,   1.66496011e-03,   3.32992022e-03, ...,
         2.39704307e+01,   2.39720957e+01,   2.39737606e+01])
(Pdb) raw.time_as_index(ts)
array([    0,     1,     2, ..., 14397, 14397, 14399])
```

Notice the non-unique index values.

wronk, 2015-07-17:
I'll take care of this one later, @mainakjas and @dengemann wouldn't assign it even after asking politely.

mainakjas, 2015-07-17:
you are @wronk @wronk

Eric89GXL, 2015-07-17:
We've actually had this discussion before at some point, I think we have to
keep the truncation behavior otherwise it ends up acting like an API
change. I agree it can produce some unexpected results, but it might be too
late to change it.

wronk, 2015-07-19:
So it would just cause too many problems downstream?

Eric89GXL, 2015-07-19:
We can't know what the effect would be on people's pipelines, yeah. Ideally
stich a small change would not matter in practice, but we can't guarantee
that.

wronk, 2015-07-19:
Could we add an 'new_behavior=False' flag and add an option to use the corrected indices?

agramfort, 2015-07-19:
I am fine with adding a round=True or False option to time_as_index

wronk, 2015-07-24:
PR #2350

teonlamont, 2015-07-28:
closed for #2350

closed 2015-07-28 by teonlamont