mne-tools/mne-python #2315: ENH baseline argument to read_epochs

teonlamont, 2015-07-17:
This was brought up on the listserv. It would be useful to have a baseline argument loading epochs with `read_epochs`. I think that the internals are there, they just need to be exposed.

agramfort, 2015-07-19:
why not however baseline estimates on single trials are really noisy.

It is an easy PR. It's basically adding a baseline param to read_epochs as
it is with read_evokeds

jona-sassenhagen, 2015-07-21:
http://www.ncbi.nlm.nih.gov/pmc/articles/PMC3183439/ a paper from the EEGLAB people (including Delorme) arguing that for time-frequency decompositions, single-trial baselining is better. Doubt it translates directly to the time domain though ... especially with the very short windows most people seemingly like.

Also Teon if you find yourself wanting to do this, can you add a baseline method to Evoked while you're at it? rERPs skip the Epoch step and for now need to be baselined manually (e.g. with mne.baseline.rescale).

agramfort, 2015-07-21:
it's a problem of spectral leakage and circular convolutions when using
FFTs. If it's ok for your analysis you should high pass filter your data a
bit.

teonlamont, 2015-07-22:
@jona-sassenhagen, i'll take care of this. @agramfort, isn't that what is being done when people generate the Epochs object and baseline correct? should we have general method for mne instances? also, should baselining be restricted to being applied only once? i presume yes.

agramfort, 2015-07-23:
> @jona-sassenhagen, i'll take care of this. @agramfort, isn't that what is being done when people generate the Epochs object and baseline correct? should we have general method for mne instances?

we could.

> also, should baselining be restricted to being applied only once? i presume yes.

not needed except that you cannot undo basline. But you could do

evoked.apply_baseline((None, 0))

then change basline:

evoked.apply_baseline((-0.1, 0))

and go back to previous baseline with:
evoked.apply_baseline((None, 0))

Eric89GXL, 2015-07-23:
Yeah, as long as we throw an error for `apply_baseline(None)` we should be fine to change repeatedly

closed 2015-09-10 by Eric89GXL