mne-tools/mne-python #2316: ENH ICA plot_sources

teonlamont, 2015-07-17:
@jaeilepp is there to recycle `topo` figures in the `ica.plot_sources()`? For instance, if I click the same IC twice, it generates two figures.

jaeilepp, 2015-07-20:
What do you mean by recycling? You mean that each channel is in separate
figure? That should be easy to do...

teonlamont, 2015-07-20:
actually, that's the current behavior already. what I am suggesting is that when the first figure for an IC is generated, it creates a reference to that figure, so if you click on it again, instead of creating a new figure, it just brings that one back. it would only create a new copy of the topo if it were previously closed.

agramfort, 2015-07-20:
please keep it simple.

teonlamont, 2015-09-08:
closing for now

closed 2015-09-08 by teonlamont