mne-tools/mne-python #2317: ENH: ROI plot

jona-sassenhagen, 2015-07-18:
http://nbviewer.ipython.org/gist/jona-sassenhagen/ca7d95d2dbfd8a5a0bd7

This could eventually become mne.viz.evoked.plot_roi(...).

I'm not opening a PR for now because I have way too many open already (please merge #2304 :) ... I'll close #1890 soon).

Addresses parts of #2268 (see auto roi assignment helper functions) and #1939. @teonlamont

agramfort, 2015-07-19:
can you explain how it works? It's really justified for MEG to aggregate
channels due to orientation of sensors btw.

Thanks for keeping the number of PRs under control :)

jona-sassenhagen, 2015-07-23:
The idea is, in EEG research, you often present averages over multiple sets of channels - e.g., you'd present one plot for occipital, one for frontal electrodes and so on. This is a function that either

- takes a dict kwarg specifying ROI assignments of electrodes
or
- automatically assigns electrodes to ROIs based on spatial neighborhood

and then plots the mean of all sensors for each ROI, per condition.

The aux functions can also simply be used to create ROI mappings, which could then be improved by hand.

If I understand Teon correctly, for MEG channel pairs, you can take the RMS (or possibly the difference, or the abs values?) of precisely overlapping sensors, but once you've done that, you can average over these RMS'd pairs? I tried googling it, but didn't find anything my poor EEG mind could understand.

dengemann, 2015-07-23:
> can you explain how it works? It's really justified for MEG to aggregate
channels due to orientation of sensors btw.

it's not really justified.

I'm a bit reluctant of supporting this. I'm loosing the overwview about too many PRs and viz features that kind of do the same. Could someone who needs this kind of funcitonality please propose a concise and efficient API cleanup. I'm a bit affraid that we push carelessly incremental stuff which will burn a lot of time.

teonlamont, 2015-07-23:
for the MEG rois, we should look at mne-c. I believe @mshamalainen had select ROIs implemented for sensor-space. we could see how it's done there.

agramfort, 2015-07-24:
yes it is in mne/data/mne_analyze.sel but it's selection only of neuromag
by default.

jona-sassenhagen, 2015-07-27:
Can I get a vote/opinion if something like this would be welcome in MNE?
It's not how I'd personally visualise data, but friends wanted a visualisation technique like this so I wrote it.

agramfort, 2015-07-28:
Mixed feeling on my side... I would not use this either.

jona-sassenhagen, 2015-07-29:
As an example for this kind of plot from the literature, see e.g. http://www.ncbi.nlm.nih.gov/pmc/articles/PMC4488605/figure/F3/

And yes, it's quite straight-forward to make a plot like this by hand.

But really, if you guys don't want it, just tell me and I'll close this.

kingjr, 2015-07-29:
I'll probably use it for EEG datasets.

agramfort, 2015-07-29:
maybe you can make a tiny example with matplotlib so we don't face any
maintenance burden for a feature that does not seem core to me.

ImmanuelSamuel, 2015-08-07:
For people using high-density EEG (128, 256 channels), providing a 3X3 channel representation that shows a generalized effect is very useful.

agramfort, 2015-08-07:
I would write an example then but don't add a specific function in the MNE
code base. It's fine if example is a bit long. At least we'll have a demo
and we don't need to worry much about maintenance.

Eric89GXL, 2015-10-19:
@jona-sassenhagen would it be time consuming to open a WIP PR with an example to see if it satisfies your use case, @teonlamont's use case from #1939, and @ImmanuelSamuel's use case? It seems like there are enough people looking for ROI-type plots that it could be useful. It'll have to be for 0.11+, though.

jona-sassenhagen, 2015-10-19:
It should be fairly low effort, but I do have 3 open right now (though one is at merge status).

Eric89GXL, 2015-10-19:
Great, feel free to tackle it whenever you get a chance, then. No rush since I think we'll want a release cycle (0.11) to work it out.

jona-sassenhagen, 2015-11-26:
close for #2602

closed 2015-11-26 by jona-sassenhagen