mne-tools/mne-python #2321: ENH: Disable gaussian smoothing in mne.viz.plot_topo_image_epochs

leggitta, 2015-07-20:
Hello,

I'd like to disable the gaussian smoothing when plotting image epochs, so I can clearly identify which trials have artifacts in them. I thought that the user could set ```sigma=None```, in which case the smoothing step is skipped. Would anyone else be interested in this option?

Alan

agramfort, 2015-07-20:
what if you set sigma=0. ? does it do the job?

leggitta, 2015-07-20:
This is with sigma=0, but I was hoping for something more pixellated so I can see where the edges are.

[image]

Eric89GXL, 2015-07-20:
What does it look like with nearest-neighbor interpolation (instead of bilinear) on the `imshow` with different sigmas? It would make sense to me to have NN interp on by default, since the sigmas should take care of the smoothing for you.

leggitta, 2015-07-20:
Here's the same image with nearest

[image]

agramfort, 2015-07-20:
looks good no?

Eric89GXL, 2015-07-20:
That is what I would expect from `sigma=0.` and nearest-neighbor interpolation. The smearing on the temporal axis is due to the overlapping of the windows used.

What does it look like for the default sigma value? Is it very different from how it is with bilinear interp on?

leggitta, 2015-07-20:
Ok, here it is with ```interpolation='nearest'``` with the default sigma value.

[image]

leggitta, 2015-07-20:
so the smoothing is happening in the temporal domain, not across trials?

agramfort, 2015-07-20:
It should happen across trials. We have a bug???

leggitta, 2015-07-20:
It is smoothing across trials, I just tried out a few different sigmas. But if I wanted to look at the trials without blurring, I need to change both sigma and the interpolation argument.

Eric89GXL, 2015-07-20:
try sigma=2, it should make it smooth vertically

Maybe we should change the default to something like that. I think in any case we should set the interpolation to nearest. It is a bit silly to have two forms of interpolation going on like that.

agramfort, 2015-07-21:
+1 for smoothing vertically and use nearest.

volunteer?

Eric89GXL, 2015-07-21:
@leggitta you find that it's broken you buy it? :)

leggitta, 2015-07-21:
Sounds fair. The vertical smoothing seems to be working fine. The below plot is with sigma=2 and interpolation='nearest'.

Should I make interpolation='nearest' for all functions that use imshow?

[image]

Eric89GXL, 2015-07-21:
The smoothing type probably needs to be decided on a case-by-case basis. Are there other ones where you think it would make sense?

leggitta, 2015-07-21:
Not of the top of my head, but I'll look around for ones that also apply some kind of gaussian smoothing.

PR #2325 (2015-07-22): No smooth
Sets ```interpolation='nearest'``` for plotting image epochs (since these use gaussian filtering). 

resolve #2321
closed 2015-07-21 by leggitta