mne-tools/mne-python #2324: bug in mne_browse_epochs

mainakjas, 2015-07-21:
@jaeilepp if I click on the lower right corner of the epochs browser (below the scroll bar), I get an error which says:

```py
In [2]: Floating point exception
```
Ipython closes and everything crashes. Maybe I should do more science to discover such corner cases ;)

mainakjas, 2015-07-21:
I think the scrollbar has to be at the bottommost position for this error to occur. So, scroll to the bottom and then click on the bottom left corner. @yousrabk couldn't reproduce this on her computer but I can reproduce it on the example. If anyone else can reproduce it, maybe that's useful info

jaeilepp, 2015-07-24:
I can't reproduce.

mainakjas, 2015-10-02:
I can show you on my computer :)

Eric89GXL, 2015-11-25:
@jasmainak you still having this issue?

jasmainak, 2015-11-25:
yes, this is not fixed and @jaeilepp has been able to reproduce it on my computer

jaeilepp, 2016-06-23:
This is solved. Closing...

closed 2016-06-23 by jaeilepp