mne-tools/mne-python #2330: implement draw_sensors

agramfort, 2015-07-22:
implement draw_sensors

https://github.com/mne-tools/mne-matlab/blob/master/matlab/mne_load_coil_def.m#L160

in plot_trans function.

Eric89GXL, 2015-07-22:
We have to be careful because that was translated from BrainStorm code, which I think is GPL'ed. So we'll have to ask permission to relicense, right @agramfort?

agramfort, 2015-07-22:
hum... @ftadel what do you say?

ftadel, 2015-07-22:
I don't think we can even trace back who wrote that function...
I think you can just re-use whatever you need to :)

agramfort, 2015-07-23:
ok cool.

any volunteer?

Famguy, 2016-01-05:
@agramfort I would like to volunteer for this (in case nobody's working on it).

agramfort, 2016-01-05:
go for it.

#2599 or #2598 might be a little easier though

jasmainak, 2016-01-06:
This one might be easy too: https://github.com/mne-tools/mne-python/issues/2770

PR #3030 (2016-03-21): MRG: Draw sensors
So far I've modified the example and it looks good.

Before:

[image]

After:

[image]

I'm thinking of adding to the example to plot layouts from other systems since we have those now. Think it's worthwhile? I guess I should at least beef up the tests so that our coverage doesn't take a hit.

Closes #2330.
closed 2016-03-21 by Eric89GXL