mne-tools/mne-python #2331: Optimize linear_regression_raw: sparse matrices, partial fits

jona-sassenhagen, 2015-07-23:
Currently, either many predictors, or long time windows, or high sampling rates w/o decim, or especially any combination of these can lead to very memory-intensive models in rER[F/P]s that overload even powerful computers, and laptops may struggle even with smaller problems.

In response,
- predictor matrix X should be made sparse
- fitting can be done iteratively/on non-overlapping subsets

See #2304, especially @njsmith's last comment (Thanks for the insight!).

I will try, unless somebody better at numpy/scipy/linear algebra wants to.

mainakjas, 2015-07-23:
I think the first step is to profile, to figure out what needs to be optimized.

jona-sassenhagen, 2015-07-23:
Yup, edited. Though really, creating, and then inverting, the pred matrix is definitely the culprit of at least most of the problems, I've already found that while writing the function.

I actually managed to overload our 256GB server a few times by accidentally giving a transpose of a matrix, leading to a gigantic inversion.

dengemann, 2015-07-23:
However, +1 for profiling

jona-sassenhagen, 2015-08-06:
Even though iterative fitting isn't done yet, maybe we could for now close this for #2382 as it has brought decent performance improvements and I'd rather do other stuff before turning to iterative fitting.

agramfort, 2015-08-06:
Fine with me

jona-sassenhagen, 2015-08-06:
close for #2382 for now

closed 2015-08-06 by jona-sassenhagen