mne-tools/mne-python #2333: ISSUE? ica.plot_components() requires ch_type when picks is not None

jona-sassenhagen, 2015-07-23:
With my Brainvision data, these work:
```ica.plot_components()```
```ica.plot_components(picks=[1,2], ch_type="eeg")```

but this doesn't:
```ica.plot_components(picks=[1,2])```
```

---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-16-64eeca493160> in <module>()
----> 1 ica.plot_components(picks=[1,2])

/home/FB/fb05/sassenha/tools/mne-working/mne-python/mne/preprocessing/ica.py in plot_components(self, picks, ch_type, res, layout, vmin, vmax, cmap, sensors, colorbar, title, show, outlines, contours, image_interp, head_pos)
   1352                                    outlines=outlines, contours=contours,
   1353                                    image_interp=image_interp,
-> 1354                                    head_pos=head_pos)
   1355 
   1356     def plot_sources(self, inst, picks=None, exclude=None, start=None,

/home/FB/fb05/sassenha/tools/mne-working/mne-python/mne/viz/topomap.py in plot_ica_components(ica, picks, ch_type, res, layout, vmin, vmax, cmap, sensors, colorbar, title, show, outlines, contours, image_interp, head_pos)
    690 
    691     data_picks, pos, merge_grads, names, _ = _prepare_topo_plot(ica, ch_type,
--> 692                                                                 layout)
    693     pos, outlines = _check_outlines(pos, outlines, head_pos)
    694     if outlines not in (None, 'head'):

/home/FB/fb05/sassenha/tools/mne-working/mne-python/mne/viz/topomap.py in _prepare_topo_plot(inst, ch_type, layout)
     56 
     57         if len(picks) == 0:
---> 58             raise ValueError("No channels of type %r" % ch_type)
     59 
     60         if layout is None:

ValueError: No channels of type 'mag'
```

Am I missing something, or is this a bug?

dengemann, 2015-07-23:
This looks like a bug. Btw. didn't we ant to guess the channel type at some point? Why do we still need to pass the channel type? @wmvanvliet @mainakjas @agramfort

jona-sassenhagen, 2015-07-24:
Closed for my bad PR at #2336 I guess.

closed 2015-07-24 by jona-sassenhagen