mne-tools/mne-python #2341: Explained variance of PCs

teonlamont, 2015-07-23:
Is there an easy way to get the percentage of explained variance for each of the principal components?

agramfort, 2015-07-24:
no but it would indeed be a useful infomation to store in a proj object.

teonlamont, 2015-09-08:
combining #1999:
>
plot_projs_topomap defaults to the Vectorview-all if a layout isn't specified. plot_ica_components uses a neat little helper function _prepare_topo_plot to get the position information. I think this would be a good alternative, but the downside is that the proj objects don't have an info attribute. if it had this, it would make it easy for non-Vectorview systems.

agramfort, 2015-09-09:
adding an info to proj files is possible but a mess with historical files.

is passing info to the function so annoying?

dgwakeman, 2015-09-09:
why not have the proj's added to whatever you are calculating it on average, raw, etc. then just plot the projections from that object. i.e. mne.plot_proj(raw), which has the info?

agramfort, 2015-09-10:
I realize with already have a plot_projs_topomap in the ProjMixin so you
can do:

evoked.plot_projs_topomap()
epochs.plot_projs_topomap()
raw.plot_projs_topomap()

teonlamont, 2015-09-10:
ok cool no mod of `proj` for `info`. does `proj` store the diagonal matrix anywhere, I'm assuming that its eigenvalues would give the percentage of explained variance

agramfort, 2015-09-11:
no it does not store it so far.

teonlamont, 2015-12-13:
can this be determined from the S matrix in the SVD? if I remember my linear algebra, the diagonal should contain the eigenvalues in ranked order. to get the explained variance, their sum should be scaled to one.

jasmainak, 2015-12-13:
you seem to be on the right track @teonlamont ;) PR welcome

jasmainak, 2015-12-13:
this may help you: https://github.com/scikit-learn/scikit-learn/blob/master/sklearn/decomposition/pca.py#L284 :)

teonlamont, 2015-12-14:
@jasmainak, thanks for the link.

a few questions have come up:
1. is there a reason why the [computed covariance](https://github.com/mne-tools/mne-python/blob/master/mne/proj.py#L160) isn't mean-centered?

2. from the looks of things `data` that goes into [_compute_proj](https://github.com/mne-tools/mne-python/blob/master/mne/proj.py#L71). it's already the covariance at this point, so its SVD should be `U * S**2 * Ut`. is this correct?

2. should this be a separate function of an add-on the existing ones? It easily falls out of the computation. Otherwise, the PCA will be recalculated for the explained variance function.

agramfort, 2015-12-14:
EEG and MEG data are commonly assumed to be zero mean simply because you
apply baseline correction or high pass filtering. ok?

teonlamont, 2015-12-14:
got it. opened a pr.

PR #2703 (2015-12-18): [MRG] ENH add explained_variance to PCA
closes #2341. this is written to return an optional `explained_variance`. I am not sure if this is pythonic. The other option would be to make a separate function but then it wouldn't be DRY.
closed 2015-12-18 by agramfort