mne-tools/mne-python #2344: Make ylim and scalings play nicely in Evoked.plot

teonlamont, 2015-07-23:
Currently, ylim and scaling are not playing nicely with each other. If you set 
`ylim = dict(mag=[-300e-15, 300e-15])` and scalings=None, the ylim is not adjusted for the default scaling `mag=1e15`. we should fix it so that `ylim` is in regular units but is scaled internally with the `scalings` param. other option is to explain the behavior, but I feel like no one would expect this behavior.

agramfort, 2015-07-24:
well the rationale for the current behavior is that the ylim look similar
between sensors and you don't need to think about the exponent for each
channel type

teonlamont, 2015-07-24:
oh ok, i understand the rationale. maybe I should add a note that the ylim should take the scaling into consideration.

agramfort, 2015-07-24:
Go for the note

teonlamont, 2015-09-08:
closing because it only seemed to catch me off guard

closed 2015-09-08 by teonlamont