mne-tools/mne-python #2345: Add event_id to read_epochs or selectively read

teonlamont, 2015-07-24:
If you only want to load a subset of your epochs from disk, suppplying an event_id could save on memory. currently, it loads all the epochs, and then you can do a selection. for long experiments, and large numbers of epochs, this is not idea. the event id would find the indices in the events and from there, it can index and load just those epochs. 

one thing that I am not certain of is whether all the epochs would still be temporarily stored in memory. if that's the case, is there any way to read only select epochs?

cc: @Eric89GXL

Eric89GXL, 2015-07-24:
This would be solved by the more general solution of adding general
on-access reads, so that's my preferred solution. I might be able to do it
next week.

teonlamont, 2015-09-08:
are we planning to deal with the epochs reading on demand in this cycle?

Eric89GXL, 2015-09-08:
I might be able to do it. Would that supercede this?

teonlamont, 2015-09-09:
yeah, i believe so, on-demand then event_id capability.

teonlamont, 2015-09-15:
closing for #2466

PR #2466 (2015-09-16): MRG: Allow preload=False in read_epochs
Closes #2345.

@teonlamont this should work for you.
closed 2015-09-15 by teonlamont