mne-tools/mne-python #2346: to_data_frame error with BDF/EDF raw

cle1109, 2015-07-24:
After loading a .BDF file, I get the following error when trying to convert to a data frame using `to_data_frame`:

    ---------------------------------------------------------------------------
    TypeError                                 Traceback (most recent call last)
    <ipython-input-158-eb57e0ff561f> in <module>()
    ----> 1 df = raw.to_data_frame()

    /usr/local/lib/python3.4/site-packages/mne/io/base.py in to_data_frame(self, picks, index, scale_time, scalings, copy, start, stop)
        157                 col_names = [self.ch_names[k] for k in picks]
        158 
    --> 159             types = [channel_type(self.info, idx) for idx in picks]
        160             n_channel_types = 0
        161             ch_types_used = []

    TypeError: 'NoneType' object is not iterable

The deprecated method `as_data_frame` works as expected.

Eric89GXL, 2015-07-24:
@choldgraf sounds like a problem with the exporter, probably because `picks=None`?

choldgraf, 2015-07-24:
Ah probably yes - how does MNE usually handle this? Just something like:

```python
if picks is None:
  picks = range(self.n_channels)
```
?

agramfort, 2015-07-24:
picks=None means either all channels or just the data channels (meg + eeg)
depending on context.

cle1109, 2015-07-24:
I tried the following:

    df = raw.to_data_frame(picks=list(range(raw.info['nchan'])))

But this results in:

    ---------------------------------------------------------------------------
    UnboundLocalError                         Traceback (most recent call last)
    <ipython-input-170-393334cf5c9a> in <module>()
    ----> 1 df = raw.to_data_frame(picks=list(range(raw.info['nchan'])))

    /usr/local/lib/python3.4/site-packages/mne/io/base.py in to_data_frame(self, picks, index, scale_time, scalings, copy, start, stop)
        171                 idx = [picks[i] for i in range(len(picks)) if types[i] == t]
        172                 if len(idx) > 0:
    --> 173                     data[:, idx] *= scaling
        174 
        175         # Make sure that the time index is scaled correctly

    UnboundLocalError: local variable 'data' referenced before assignment

choldgraf, 2015-07-24:
ah ok it looks like there's a self._get_check_picks function but for some data types it is not being called...I will fix this soon but my computer is about to die

agramfort, 2015-07-24:
PRs welcome :)

choldgraf, 2015-07-24:
Will do when my computer is no longer dead.

agramfort, 2015-07-24:
let me guess. You tried the linear_regression_raw and it freezed your
laptop? :)

choldgraf, 2015-07-24:
Haha I was having too much fun at the anticafe :-)

choldgraf, 2015-07-25:
Hey -  can you try this again with the latest MNE master @cle1109? I think the issue was that to_data_frame was checking to see if the object was Raw or RawArray, when instead it should have been checking for _BaseRaw.

cle1109, 2015-07-27:
@choldgraf yes, `to_data_frame()` works now! Thanks for the fix.

closed 2015-07-27 by cle1109