mne-tools/mne-python #2347: Scaling of EEG data not 10e-6?

cle1109, 2015-07-24:
In my EEG data, I get inconsistent results when I look at the raw data in two ways:

1. I access the data directly using `eeg, _ = raw[:, :]`.
2. I convert to a data_frame using `df = raw.as_data_frame()`.

Now when I compare the actual data values, I get a ratio of 10e5. This is somewhat unexpected, because I thought the original EEG data is measured in volts and gets converted to microvolts (i.e. 10e-6) when converting to a data frame.

agramfort, 2015-07-24:
have a look at the as_data_frame code. It's not much used so likely to be a
bit broken.

cle1109, 2015-07-24:
Nevermind, I accidentally used `10e6` instead of `1e6`, so everything is fine :-).

closed 2015-07-24 by cle1109