mne-tools/mne-python #2349: event.py define_target_events error

natalieklein, 2015-07-24:
I brought this up on mne_analysis_request mailing list, then looked at the code, and wasn't sure the best way to point out the issue, so here it is:

Line 141:         lag[lag != fill_na] *= tsample

If fill_na is None, then lag != fill_na returns True, so this only multiplies lag[1] by tsample, leaving all other entries in lag as number of samples rather than number of milliseconds. Not sure the most elegant way to fix it - maybe just if fill_na is None, do lag *= tsample, else do the current line 141. But it also seems risky to index this way; what if the true lag is 47 and my fill_na is 47 as well?

Thanks,

Natalie

agramfort, 2015-07-25:
can you paste here 10 lines of python that reproduces the bug and that one
could use for a test?

thanks for looking into it.

natalieklein, 2015-07-27:
This code reproduces the bug - you can see it only multiplies lag[1] by tsample rather than all non-NA elements by inspecting new_events[1], as all of these lags should be exactly 1500ms:

events = np.array([[0, 0, 1], [375, 0, 2], [500, 0, 1], [875, 0, 3], [1000, 0, 1], [1375, 0, 3], [1100, 0, 1], [1475, 0, 2], [1500, 0, 1], [1875, 0, 2]])                              

new_events = mne.event.define_target_events(events=events, reference_id=1, target_id=2, sfreq=250., tmin=1.4, tmax=1.6, new_id=5)

Thank you!

dengemann, 2015-07-27:
@natalieklein sorry for the delay, I'm currently traveling, I'll get back to you ASAP.

natalieklein, 2015-07-27:
@dengemann No problem! I used the code from your function without the lag for now as I don't really need the lag, so no rush at all, I simply wanted to share so it could be fixed for other users. I mentioned this in the mne analysis list, but as a reminder, this problem only occurs when fill_na is None.

Thanks again!

agramfort, 2015-07-28:
@natalieklein do you think you can open a PR to fix this? It should come with a non-regression test that you already have here.

natalieklein, 2015-07-29:
@agramfort I'll try to do it today!

natalieklein, 2015-07-29:
@agramfort I believe I fixed the problem, and want to check how to incorporate the test before doing a PR. I haven't really contributed to a project like this before so I apologize if this is obvious, but would I just add the test to mne/test/test_events.py test_define_events() function?

PR #2364 (2015-08-20): Bugfix issue 2349
closes #2349  

I wanted all lags to multiply by tsample and to guard against the possibility that a lag would equal fill_na, which would cause unexpected behavior. 

What I did: If fill_na is not None, I append None to lag at the "na event" indices. Then np.abs turns these values to nan, so the returned lag list contains millisecond lag values for the found events and nan for the "na events".

I also attempted to add my test case to mne/test/test_events.py in the test_define_events function, but if this was the incorrect way to incorporate the test, please let me know!

Than [... cut]
closed 2015-08-20 by dengemann