mne-tools/mne-python #2357: BUG: problem with crop / decimation

kingjr, 2015-07-28:
I'm encountering some weird behaviors, but I'm unclear what they're due to:

```
import numpy as np
from mne.epochs import concatenate_epochs, EpochsArray
from mne.io.meas_info import create_info
from mne.decoding import GeneralizationAcrossTime

def make_epochs():
    n_trials = 100
    data = np.random.randn(n_trials, 64, 1946)
    times = np.linspace(-1.8, 2., data.shape[2])
    sfreq = len(times) / np.ptp([times[0], times[-1]])
    chan_names = ['eeg_%s' % chan for chan in range(64)]
    chan_types = ['eeg' for ii in range(64)]
    events = np.hstack((np.zeros(n_trials / 2), np.ones(n_trials / 2)))
    events = np.c_[np.cumsum(np.ones_like(events)) * sfreq, events, events]
    info = create_info(chan_names, sfreq, chan_types)
    epochs = EpochsArray(data, info, events=events, tmin=-1.8,
                         event_id=dict(a=0, b=1), verbose=False)
    return epochs

# BUG 1
epochs = make_epochs()
len(epochs)  # ok
epochs_ = concatenate_epochs((epochs['a'], epochs['b']))
len(epochs_)  # why not ok??
```

```
# BUG 2
gat = GeneralizationAcrossTime()
epochs = make_epochs()
epochs.crop(-.200, .700)
gat.fit(epochs)  # ok

epochs = make_epochs()
epochs.decimate(10)
gat.fit(epochs)  # ok

epochs = make_epochs()
epochs.decimate(10)
epochs.crop(-.200, .700)
gat.fit(epochs)  # ok

epochs = make_epochs()
epochs.crop(-.200, .700)
epochs.decimate(10)
gat.fit(epochs)  # why not ok??

```

Eric89GXL, 2015-07-28:
What erros do you get? Is this related to #2356?

I agree these should be doable, and made into tests to make sure everything is working properly.

kingjr, 2015-07-28:
> What erros do you get?

I update the post.

> Is this related to #2356?

I'm not sure, but I think these are separate issues.

Eric89GXL, 2015-07-28:
First bug -- it looks like somehow the concatenation function is not recognizing/passing on that bad epochs have been dropped, and probably needs to set a parameter appropriately.

If the splits are in time, that last one could actually be due to time rounding/indexing errors, or it could be something different.

agramfort, 2015-07-29:
fixed first bug in :

https://github.com/mne-tools/mne-python/commit/1cc037e71d723350f3beb54f9a545d56cc4e6683

for the second one it's something in GAT I suspect.

kingjr, 2015-07-30:
thanks @agramfort! For the second bug, it's actually a crop / decimate issue too: 
```
epochs = make_epochs()
epochs.crop(-.200, .700)
times = [epochs.times[-1]]
epochs.decimate(10)
times.append(epochs.times[-1])
print times
```
gives
```
[0.6990750256937307, 1.9878725590955804]
```
I guess you should probably update _raw_times, or compute the decimated time differently
epochs.times = epochs._raw_times[epochs._decim_slice].

I'm not sure what _raw_times is, so I'll let you fix the bug ;)

Eric89GXL, 2015-07-30:
I can look into it tomorrow

PR #2367 (2015-07-30): MRG: Fix for epochs _raw_times
Closes #2357.

Ready for review/merge from my end.
closed 2015-07-30 by agramfort