mne-tools/mne-python #2362: ENH: Patterns visualization in GeneralizationAcrossTime

trachelr, 2015-07-29:
Hey guys @kingjr @agramfort @dengemann 
I just had a quick look at the patterns obtained in the time generalization example.
There's some interesting things that explain why sometimes the decoding accuracy is below chance level. For instance, train times 50ms and 350ms give below chance performance at test time 250ms and 150ms, respectively. 
[image]
Looking at the topomaps, we can see a patterns inversion between these train and test times.
[image]
For me, this observation explain below chance performance and make sense actually.
Do you think it would be useful to create an example of these plots?

Romain

jona-sassenhagen, 2015-07-29:
This is less about the implementation of the method in the context of MNE, and more about the nature of the procedure and the interpretation of its application to real neural data, isn't it?

trachelr, 2015-07-29:
@jona-sassenhagen yess... @kingjr told me that reviewer often asked about how to provide interpretation of these decoding maps. This sign flip explain clearly why there's some poor performance regions, which is because the pattern of activity is inverted between the train and test set. Therefore the decoder prediction are also inverted.

jona-sassenhagen, 2015-07-29:
FWIW this is discussed in Jean-Remi's PLoS ONE paper.

kingjr, 2015-07-29:
Yes, this is discussed both in the PLoS ONE (where the reversal is very strong) and the TICS papers. Note the strict equivalence between negative scores and topography inversion, which allows you to make reverse inference if you see one or the other.

What's interesting in the PLoS as well as in the current example (although here we'd need more subjects to be sure), is that the reversal isn't complete: you can decode better at t and at t' respectively than you can generalize from t to t' and vice versa. This means that part of the signal is indeed signed flipped, but that this flip isn't the whole story: there's, on top of this flip, a sequence of distinct brain responses.

I agree with jona that this is a bit outside mne-python's scope. Except if you show us that the topographies based on the cov * coef are different from the univariate analyses.

PS: I think you have a unit problem in your plot.

jona-sassenhagen, 2015-07-29:
>cov * coef

... what coefficients?

And in a possibly related question - do you have a logistic regression classifier pipeline for GAT lying around that you've found works well?

kingjr, 2015-07-29:
> ... what coefficients?

the clf.coef_, to see what topographies the clf is focusing onto.

> And in a possibly related question - do you have a logistic regression classifier pipeline for GAT lying around that you've found works well?

For 2class problem with 306 sensors I use this by default: https://gist.github.com/kingjr/2476f00a28bc4d7873b7
If you have smaller dimensionality and higher SNR (e.g. if you use meta trials, reduce C to something like C=1e-4)

jona-sassenhagen, 2015-07-29:
Thanks @kingjr !

trachelr, 2015-07-29:
yep that's equivalent to univariate analyses... it's just another way of watching at the data.
@kingjr, unit problem is due to standard scaling.

kingjr, 2015-07-29:
> yep that's equivalent to univariate analyses... it's just another way of watching at the data.

Is this done with your 'plot_patterns ' PR? If so, it could be slightly different to univariate (the topographies would  could berobust to some outlier for instance). I think we should make a notebook with all the decoding and GAT ideas and interpretations, as opposed to additional examples or scripts in mne.

trachelr, 2015-07-29:
Yep it is done with the plot_patterns method in class mne.decoding.base.LinearModel.
and the patterns_ attribute is computed by cov * clf.coef_

kingjr, 2015-07-29:
> Yep it is done with the plot_patterns method in class mne.decoding.base.LinearModel.
and the patterns_ attribute is computed by cov * clf.coef_

So, if you apply a robust covariance method, then you should be able to find cases were cov * coef will be better than the univariate :). This should be advantageous 1) when outlier epochs are present in the data set and 2) when some channels have intermittent artefacts. It'd be nice because these are two types of artefacts that are a bit annoying to cleanup, so bypassing them would definitely be a plus. @dengemann do you agree?

Eric89GXL, 2017-03-21:
This sounds like a writing not coding issue (feel free to correct me), closing.

closed 2017-03-21 by Eric89GXL