mne-tools/mne-python #2365: mne.channels.read_montage doesn't work for .elp

haribharadwaj, 2015-07-29:
A collaborator asked for help reading bdf files and locations as elp files with mne-python.. 
* Specifying ```read_montage(..., path='/blah')``` doesn't work.. Only specifying the ```kind``` input with full absolute path works... Can't figure out why from code..

* The ```numpy.readtxt(.)``` call within ```read_montage(.) ``` seems to error out for a file that I was given that readlocs(.) in EEG lab is able to read without any issues..

The offending file is here: http://martinos.org/~hari/problemfile.elp

I have no experience with .elp files nor the time .. hence starting the issue :P

teonlamont, 2015-07-29:
two things i noticed briefly:
1. this `*.elp` is different than the `*.elp` that is supported (looks like the supported one is BESA spherical)
2. well, there's this https://github.com/mne-tools/mne-python/pull/1001 and it is the format of the `*.elp` that you are mentioning but this pr was for electrode points that would be used for the `DigMontage`.

The dilemma is the filename extension seems to be shared across two different formats (http://sccn.ucsd.edu/eeglab/allfunctions/readlocs.html) but we don't have a parameter to disambiguate them or support for the second case :/

agramfort, 2015-07-29:
is there a way to guess the file format?

haribharadwaj, 2015-07-29:
Ah ok... Didn't know about multiple .elp formats... Thanks @teonlamont

haribharadwaj, 2015-07-29:
Also, using EEGlab's writelocs() to save this as a BESA-spherical EEG and then trying read_montage(), it works fine :)... 

Think this issue can be closed for now...

closed 2015-07-29 by agramfort