mne-tools/mne-python #2371: Should pick_channels throw an error if you give a string?

choldgraf, 2015-07-31:
I say this because it seems to assume that you're passing it an iterable, so if you pass a string then it will break it up into individual letters, which accidentally caused some problems for me earlier.

Another strange behavior is that it will not throw an error if you supply incorrect channel names, so long as at least one of the channel names supplied does match.

E.g., two examples of strangeness:

1. If my actual channel names are ['G1', 'G2', 'G3'] and I supply ['G1', 'MNERULES', 'YESITDOES'] as the channels, then it simply ignores the last two values and returns a `Raw` object with only 'G1' pulled out.

2. If my actual channels names are ['1', '2', '3'] and I supply 'ISTHISA2CHANNELNAME' to pick_channels (note it's a string not a list), then it will silently return a Raw copy with '2' as the picked channel.

It seems like one, or both, of these behaviors is undesirable, no?

agramfort, 2015-07-31:
PR welcome to check that we get a list as input and not a str

choldgraf, 2015-07-31:
OK cool. Will do.

Sent from my phone, apologies for the curtness and type-os

Eric89GXL, 2015-11-25:
this one's done

closed 2015-11-25 by Eric89GXL