mne-tools/mne-python #2379: equalize_epoch_count uses lots of RAM

jona-sassenhagen, 2015-08-03:
```mne.epochs.equalize_epoch_counts``` (using "mintime") takes 90GB of RAM for a single subject on some of our data sets, and is really slow. However, the "truncate" method seems undesirable. If ```mne.epochs.equalize_epoch_counts``` can't be optimized trivially, how about something like this?

```
larger, smaller = ((ep1, ep2) if len(ep1.events) > len(ep2.events) else (ep2, ep1))

from copy import deepcopy
arr = deepcopy(larger.events)

tts = []
for t in np.random.permutation(smaller.events[:,0]):
    idx = (np.abs(arr[:,0]-t)).argmin()
    tts.append(arr[idx,0])
    arr = np.delete(arr, idx, axis=0)

larger.drop_epochs(np.in1d(larger.events[:,0], tts) == False, 
                   reason='EQUALIZED_COUNT')
```

Eric89GXL, 2015-08-03:
I don't think that the actual choice of the subset of epochs to use should affect the memory. If what you write above takes less memory, then we should indeed just optimize the existing method.

FWIW there is a principled reason behind the mintime. If noise properties vary as a function of time, you're selecting epochs as close in time as possible to minimize any effect of those changes.

jona-sassenhagen, 2015-08-03:
I understand that, which is why I don't just use "truncate".

My code snippet should get epoch subsets that are at least somewhat similar, while at least for my test data being much faster. Maybe simply offer it as a 3rd method option ("mintime", "truncate", "approx_mintime")?

Eric89GXL, 2015-08-03:
It seems like a better solution is to make the existing method more efficient. Have you looked to see why the existing one does not work as efficiently?

jona-sassenhagen, 2015-08-03:
I tried ... I think it's just doing a lot of iterations by doing an all-vs-all check.
For my data, ``` _get_drop_indices``` is what takes so long, and it calls ```_minimize_time_diff```once per event, which in turn "# Check[s] every possible removal to see if it minimizes". The minimization in turn relies on np.interp. Too much maths + recursion for my feeble lib arts mind.

Eric89GXL, 2015-08-03:
How long does it take? Are you sure it's the searching step that takes 90 GB? That's a *ton* of memory for such a simple operation...

Eric89GXL, 2015-08-03:
Can you save two problematic event lists to an npz and upload it somewhere? I can take a crack at optimizing it.

jona-sassenhagen, 2015-08-03:
I think I misunderstood something - the *python process* takes up a lot of RAM while this function is running, but when I run just the _get_drop_indices function in isolation on a fresh instance, it just takes up a lot of processor power for a few seconds. Seems the RAM usage is independent of the calculation.

Either way, here are two event sets taking surprisingly long on our machines.
https://www.dropbox.com/s/sb0klwra1z6v04u/ep1.npy?dl=0
https://www.dropbox.com/s/l25rqj769ahtqmp/ep2.npy?dl=0

jona-sassenhagen, 2015-08-03:
And there does actually seem to be a decent benefit from the more involved calculation; here is "mintime" vs. my method:
[image]
[image]

Eric89GXL, 2015-08-03:
What do you mean by surprisingly long? A few seconds doesn't seem bad to me, but maybe you're doing hundreds or thousands for some application that makes it painful?

Eric89GXL, 2015-08-03:
(thanks for the files, I'll try to take a look today or tomorrow)

Eric89GXL, 2015-08-03:
BTW for future reference you can use `np.savez_compressed` to save multiple arrays to a single file

kingjr, 2015-08-03:
I'm not sure this is a relevant comment, but if you're encountering this
issue because you want to avoid biased decoding predictions, have you tried
using 'class_weight' instead of data truncation ? It should be better
because you don't drop data.

jona-sassenhagen, 2015-08-03:
Indeed @kingjr - that is, 

```
from mne.decoding.time_gen import GeneralizationAcrossTime as GAT
from sklearn.preprocessing import StandardScaler
from sklearn.linear_model import LogisticRegression
from sklearn.pipeline import Pipeline
scaler = StandardScaler()
estimator = LogisticRegression(class_weight="auto")
clf = Pipeline([('scaler', scaler), ('estimator', estimator)])
gat = GAT(clf=clf)
```

something like this?

kingjr, 2015-08-03:
Yes. Most linear classifiers have this option, although they don't solve it
in the same way.

If you have a complex dataset (e.g hierarchical classes) the best is to use
sample_weight on the true class (and adapt the stratfied folding using the
latter too) which can be distinct from the fitting class. This is a bit
tricky to use it with the gat class but I can send you some code if you
need.

PR #2380 (2015-08-04): ENH: Faster time diffs
Closes #2379.

@jona-sassenhagen this takes `_minimize_time_diff` on your sample files from 32 sec to 2.9 sec on my machine.
closed 2015-08-04 by agramfort