mne-tools/mne-python #2381: use of ~ (Stupid rant by dan)

dgwakeman, 2015-08-04:
This came up in discussion of the excellent work by wronk on sss 2374 (all links not included to avoid sidelining that work).

Feel free to ignore me, but I at least wanted to voice my thoughts (please don't take anything personally in here. these are just my thoughts on the basic design). 

The issue is that a users home directory is fundamentally the wrong place to put anything that is not user specific. All users need these lookup tables for forward solution calculation. Why have dozens of copies (one in each users home directory)? That is not the point of the home directory. That is what a lib, or share directory for a software package is for.

Why would it fail to put it in some `/share/` directory?

Which sites wouldn't it work for?

My point is the basic concept for how mne-python sets this up is wrong. ~ should be reserved for user specific only data.

Cons include: wasted space, longer calculation time for all code on first run. Potential errors by users at the same site using different `sss_cal.dat` files etc. As well as the general lack of logic in not having these things just be part of a site install.

mainakjas, 2015-08-04:
With what you describe, you'd need sudo access to install `mne-python` I guess? Currently, I like to have my mne install with anaconda without having to bug the sys-admin for anything. The fact that there is a site-wide installation shouldn't prevent individual users from using `mne-python` no?

dgwakeman, 2015-08-04:
With what you describe, you'd need sudo access to install mne-python I guess? 
     Currently, I like to have my mne install with anaconda without having to bug the 
     sys-admin for anything.

Why would you need sudo access? This directory could be anywhere the person who wants to setup the sitewide install wants to put it and set it in the sitewide config?

     The fact that there is a site-wide installation shouldn't prevent individual users from 
     using mne-python no?

I don't get how having a site-wide installation prevents individual users from using mne-python?

mainakjas, 2015-08-04:
hmm ... maybe I don't know very well how the installation works :) But at least on my workstation I don't have write access to `/share/`
`drwxr-xr-x 636 root root   20480 Aug  4 18:37 share`

So, I don't know how `easy_install mne --user` would work if there was no site-wide installation / setup already with something for mne in `/share/` directory.

dgwakeman, 2015-08-04:
Ok, to clarify I don't mean `/share/`, I mean some mne-python specific `share` directory. I should have written it better

agramfort, 2015-08-04:
@dgwakeman I understand the home issue given the space constraints at MGH in home dirs yet I agree with @Eric89GXL that we should not change default behavior just for this.

For forward computation you need write access on the folder at runtime.

Regarding cal file I would use a path in the mne config file that would point on sites to the system location. Let's avoid multiple copies of this file and let's add to the logger the file used when running maxwell_filter in python.

jona-sassenhagen, 2015-08-05:
fwiw I don't even install MNE, I just download it and do sys.path.append(...

agramfort, 2015-08-05:
works but that's a bit ugly and not recommended.

Eric89GXL, 2015-08-07:
@dgwakeman I agree in principle it would be nice if we could ship the tables with the package because of the separation of user-specific and general, as you say. We don't want to increase our package size by 100 MB by distributing the files with the package (which would be the cleanest thing to do), especially given that each user can generate the files once in a matter of a few (or tens of) seconds on first use.

There are many ways users can install mne-python, and we have no guarantee the users will have write permissions on their install directory. If they install via `anaconda`, `pip`, or `python setup.py install` they probably will, but `sudo apt-get`, `sudo pip install`, and `sudo python setup.py install` versions will lack user-level write permissions. FWIW I don't know of any python packages that assume write access to their install directory, and I assume there is a good reason for that. The only safe place I know of to store things is thus in `~`. Right now, we thus live with a 50-100 MB burden on each user, despite it not being as clean a separation of user-specific and user-general (package-level) data.

That being said, we could try to think of ways to deal with this without increasing the burden on users in other undesirable ways. For example, we currently allow setting the data paths e.g. for the `sample` dataset, we could do the same for these tables, defaulting to `~/.mne/tables` as it does now. So as a site maintainer you'd have to add an instruction to users that they should do `mne.set_config('MNE_FWD_TABLE_PATH', ...)` or so, which is about as minimal pain I can think of. By contrast, I don't think assuming that the install path is writeable is a viable solution, unfortunately.

In any case, it might still be worth documenting somewhere that 50-100 MB (or whatever the amount is) will be stored in the user's directory unless `MNE_DONTWRITE_HOME` (or whatever the var is) is set.

@dgwakeman are you up for coding the potential fix and/or the documentation?

dgwakeman, 2015-08-07:
If I can think of something that addresses this more cleanly, I will give it a whirl (though I have recently noticed I have a few open issues, which I should do first. I think I'll close this. I don't know much about the install etc. things, so perhaps I'll pester @haribharadwaj to show me how he does it here, before I take a stab at this. Thanks for the discussion!

closed 2015-08-07 by dgwakeman

Eric89GXL, 2016-01-27:
@dgwakeman it just occurred to me that a clean workaround for this would be to have an `MNE_DATA_PATH` variable that would default to `''` (meaning, use `~/.mne/` as it does now) that one could use to override where things like `~/.mne/tables` currently go. That way the only thing in `~/.mne/` would be `~/.mne/mne-python.json`, storing config vars (one of which would be `MNE_DATA_PATH`) as you prefer. I *think* it would be a pretty simple PR and shouldn't break any code, if you're feeling motivated to do it still.