mne-tools/mne-python #2383: ica.plot_sources X axis

LaetitiaG, 2015-08-05:
Hi,
I wanted to use ica.plot_sources on raw data for a specific time period, so I used the start/stop option to specify it. But whatever the start option I precise, the X scale always begins with 0. Is it just the scaling display that is wrong or is it really the data shown? Additionally, it would be nice to add the legend (to be sure that it's time in seconds). Maybe it would be easier if the start/stop options in the function were directly time in seconds (and not indices). 
Thanks,
L.

agramfort, 2015-08-05:
certainly something for @jaeilepp :)

jaeilepp, 2015-08-06:
So is the idea that start and stop should crop the data? The start and stop parameters are for selecting the time window to plot and are there basically for legacy reasons, because now it's possible to change the time window at runtime.

LaetitiaG, 2015-08-06:
Well, I'm not saying that they should (or not) crop the data, but that the X-scales legend is not in accordance with the actually displayed data window. I just want to see in the legend, for instance, 20s to 40s if I selected the start/stop indices corresponding to my data at 20s/40s. I hope I'm clear. (for the moment, the X scale legend will display something like 0 to 20)

jaeilepp, 2015-08-06:
Wait... Are you using the old version? In the current development branch the plot_sources is completely remade and looks something like this:
[image]

LaetitiaG, 2015-08-06:
Ahh, it looks like I'm using the old one (mne 0.9.0) with one plot per component. And it looks like the new one looks great! So just don't take into account my previous comments, sorry! 
When will the new version be released?

jaeilepp, 2015-08-06:
I don't know the release schedule, but the instructions for getting the latest development version can be found here: http://martinos.org/mne/stable/getting_started.html.

Eric89GXL, 2015-08-06:
Last conversation I remember we talked about a September release.

agramfort, 2015-08-06:
+1 for september.

LaetitiaG, 2015-08-06:
Ok, thanks!

closed 2015-08-06 by LaetitiaG