mne-tools/mne-python #2389: Q: topomap

ImmanuelSamuel, 2015-08-07:
I have a topo plot from eeglab and mne. They look very different. I am assuming that it is because eeglab flattens the electrode position or something. Its just that eeglab plot so much cleaner. 

EEGLAB
[image]

MNE python
[image]

ImmanuelSamuel, 2015-08-07:
data = -0.00054827592108 0.031668073094 0.0704223162929 0.0865540229014 0.085150043403 0.0457380793299 0.0509495732895 0.0987022550764 0.0801886633895 0.04976969651 0.00614093831533 -0.0373887332355 -0.0197260623702 -0.000444422313751 0.0248963507643 0.0589171499815 0.0717817674042 0.0787944587508 0.130753138626 0.124899705387 0.0975269669664 0.0419575393272 -0.00793216362559 -0.0151265531253 -0.0348081764706 0.0215552729428 0.0249141780785 0.0655540808354 0.108124467678 0.124820721722 0.146384148659 0.140718598747 0.0199164974076 0.0571570915027 0.0876578355539 0.0923016839654 0.133136594185 0.152798531971 0.122602552914 0.0714821347463 0.0537612111322 0.0469385417282 0.0740553145655 0.0915974711193 0.0743604227044 0.0110220331254 0.0234080078001 0.0325478142343 0.0477724066775 0.047792830146 0.0666905921563 0.0189050684572 0.00897804762235 0.0227890832216 0.0197423673457 -0.00454527858583 -0.00615177981404 -0.0201675075456 -0.0920040702005 -0.0464443873328 -0.0414552367577 -0.0531930278853 -0.0307912429832 -0.0412551750461 -0.0123007168696 -0.0112167955503 -0.0542948025463 -0.0857190408132 -0.0771482839 -0.113035664 -0.124491501499 -0.122871051658 -0.10429939481 -0.0749259731853 -0.0494703403965 -0.0760484386394 -0.0866986470174 -0.108261995558 -0.103236351804 -0.0826894600891 -0.114552515316 -0.0938510344287 -0.0872763177219 -0.0658120109097 -0.109361027459 -0.0545709142688 -0.0465159668016 -0.0374219872666 -0.0657419626256 -0.0692102438198 -0.0685479219719 -0.0770634783383 -0.0872426401117 -0.0666948763193 -0.0573920809363 -0.0800195158044 -0.0249282154474 -0.0257314683258 -0.0342883027284 -0.0602967350127 -0.0379301562186 -0.0422883555211 -0.0357673248702 -0.00463023885605 -0.0154207203639 -0.0236424654065 -0.00446259655858 0.00180786601789 -0.000400985259411 0.000962388449819 0.0113404811841 0.0430510059066 0.0267019564535 -0.00120974681118 0.00317540453295 0.00662862089829 -0.0460368212177 -0.0417816042596 0.00620115910578 -0.0218318617073 -0.0349000304808 -0.0231747760115 0.0320176071901 0.00236553571098 0.0207500486866 0.0235998503673 -0.00664424438941 -0.0175665711621

ImmanuelSamuel, 2015-08-07:
Please use the above data to replicate the effect

agramfort, 2015-08-07:
are you using a montage file?

ImmanuelSamuel, 2015-08-07:
Its an SFP file

     A01	       0	       0	       1
     A02	       0	-0.19937	 0.97992
     A03	       0	-0.39073	  0.9205
     A04	       0	-0.56641	 0.82413
     A05	-0.27528	-0.66458	 0.69466
     A06	-0.50865	-0.50865	 0.69466
     A07	-0.59637	-0.59637	  0.5373
     A08	-0.54874	-0.75528	 0.35837
     A09	-0.57972	-0.79792	 0.16505
     A10	-0.58743	-0.80852	-0.034899
     A11	-0.57154	-0.78666	-0.23345
     A12	-0.53271	-0.73322	-0.42262
     A13	-0.28006	-0.86195	-0.42262
     A14	-0.30048	-0.92478	-0.23345
     A15	-0.30883	-0.95048	-0.034899
     A16	-0.30478	-0.93801	 0.16505
     A17	-0.28849	-0.88789	 0.35837
     A18	-0.32275	-0.77919	  0.5373
     A19	       0	-0.71934	 0.69466
     A20	       0	-0.84339	  0.5373
     A21	       0	-0.93358	 0.35837
     A22	       0	-0.98629	 0.16505
     A23	       0	-0.99939	-0.034899
     A24	       0	-0.97237	-0.23345
     A25	       0	-0.90631	-0.42262
     A26	 0.28006	-0.86195	-0.42262
     A27	 0.30048	-0.92478	-0.23345
     A28	 0.30883	-0.95048	-0.034899
     A29	 0.30478	-0.93801	 0.16505
     A30	 0.28849	-0.88789	 0.35837
     A31	 0.32275	-0.77919	  0.5373
     A32	 0.27528	-0.66458	 0.69466
     B01	 0.18961	-0.061608	 0.97992
     B02	 0.27629	-0.27629	  0.9205
     B03	 0.50865	-0.50865	 0.69466
     B04	 0.59637	-0.59637	  0.5373
     B05	 0.54874	-0.75528	 0.35837
     B06	 0.57972	-0.79792	 0.16505
     B07	 0.58743	-0.80852	-0.034899
     B08	 0.57154	-0.78666	-0.23345
     B09	 0.53271	-0.73322	-0.42262
     B10	 0.78666	-0.57154	-0.23345
     B11	 0.80852	-0.58743	-0.034899
     B12	 0.79792	-0.57972	 0.16505
     B13	 0.75528	-0.54874	 0.35837
     B14	 0.95048	-0.30883	-0.034899
     B15	 0.93801	-0.30478	 0.16505
     B16	 0.88789	-0.28849	 0.35837
     B17	 0.77919	-0.32275	  0.5373
     B18	 0.66458	-0.27528	 0.69466
     B19	 0.49052	 -0.2832	 0.82413
     B20	 0.39073	       0	  0.9205
     B21	 0.56641	       0	 0.82413
     B22	 0.71934	       0	 0.69466
     B23	 0.84339	       0	  0.5373
     B24	 0.93358	       0	 0.35837
     B25	 0.98629	       0	 0.16505
     B26	 0.99939	       0	-0.034899
     B27	 0.95048	 0.30883	-0.034899
     B28	 0.93801	 0.30478	 0.16505
     B29	 0.88789	 0.28849	 0.35837
     B30	 0.77919	 0.32275	  0.5373
     B31	 0.66458	 0.27528	 0.69466
     B32	 0.49052	  0.2832	 0.82413
     C01	 0.11719	 0.16129	 0.97992
     C02	 0.27629	 0.27629	  0.9205
     C03	 0.50865	 0.50865	 0.69466
     C04	 0.59637	 0.59637	  0.5373
     C05	 0.75528	 0.54874	 0.35837
     C06	 0.79792	 0.57972	 0.16505
     C07	 0.80852	 0.58743	-0.034899
     C08	 0.58743	 0.80852	-0.034899
     C09	 0.57972	 0.79792	 0.16505
     C10	 0.54874	 0.75528	 0.35837
     C11	  0.2832	 0.49052	 0.82413
     C12	 0.27528	 0.66458	 0.69466
     C13	 0.32275	 0.77919	  0.5373
     C14	 0.28849	 0.88789	 0.35837
     C15	 0.30478	 0.93801	 0.16505
     C16	 0.30883	 0.95048	-0.034899
     C17	       0	 0.99939	-0.034899
     C18	       0	 0.98629	 0.16505
     C19	       0	 0.93358	 0.35837
     C20	       0	 0.84339	  0.5373
     C21	       0	 0.71934	 0.69466
     C22	       0	 0.56641	 0.82413
     C23	       0	 0.39073	  0.9205
     C24	 -0.2832	 0.49052	 0.82413
     C25	-0.27528	 0.66458	 0.69466
     C26	-0.32275	 0.77919	  0.5373
     C27	-0.28849	 0.88789	 0.35837
     C28	-0.30478	 0.93801	 0.16505
     C29	-0.30883	 0.95048	-0.034899
     C30	-0.58743	 0.80852	-0.034899
     C31	-0.57972	 0.79792	 0.16505
     C32	-0.54874	 0.75528	 0.35837
     D01	-0.11719	 0.16129	 0.97992
     D02	-0.27629	 0.27629	  0.9205
     D03	-0.50865	 0.50865	 0.69466
     D04	-0.59637	 0.59637	  0.5373
     D05	-0.75528	 0.54874	 0.35837
     D06	-0.79792	 0.57972	 0.16505
     D07	-0.80852	 0.58743	-0.034899
     D08	-0.95048	 0.30883	-0.034899
     D09	-0.93801	 0.30478	 0.16505
     D10	-0.88789	 0.28849	 0.35837
     D11	-0.77919	 0.32275	  0.5373
     D12	-0.66458	 0.27528	 0.69466
     D13	-0.49052	  0.2832	 0.82413
     D14	-0.39073	       0	  0.9205
     D15	-0.18961	-0.061608	 0.97992
     D16	-0.27629	-0.27629	  0.9205
     D17	-0.49052	 -0.2832	 0.82413
     D18	-0.56641	       0	 0.82413
     D19	-0.71934	       0	 0.69466
     D20	-0.84339	       0	  0.5373
     D21	-0.93358	       0	 0.35837
     D22	-0.98629	       0	 0.16505
     D23	-0.99939	       0	-0.034899
     D24	-0.95048	-0.30883	-0.034899
     D25	-0.93801	-0.30478	 0.16505
     D26	-0.88789	-0.28849	 0.35837
     D27	-0.77919	-0.32275	  0.5373
     D28	-0.66458	-0.27528	 0.69466
     D29	-0.75528	-0.54874	 0.35837
     D30	-0.79792	-0.57972	 0.16505
     D31	-0.80852	-0.58743	-0.034899
     D32	-0.78666	-0.57154	-0.23345

agramfort, 2015-08-07:
please share a full code snippet we can easily copy paste to reproduce the
pb.

You can use gist.github.com

ImmanuelSamuel, 2015-08-07:
https://gist.github.com/ImmanuelSamuel/39a1762374ef26888306

ImmanuelSamuel, 2015-08-07:
https://gist.github.com/39a1762374ef26888306.git

agramfort, 2015-08-07:
can an EEG friend pitch in?

cc @wmvanvliet @haribharadwaj @dengemann @christianbrodbeck ?

haribharadwaj, 2015-08-08:
@agramfort, @ImmanuelSamuel: This is just the effect of how the actual 3D xyz map on to the 2D plane of the plot... mne-python just does a projection whereas eeglab seems to do a surface transformation (i.e., like globe to map). This can be seen from how the density of dots representing electrode locations is different between the two.

In sum, it is a not a problem per se unless there is a preferance for one kind of visualization over the other... In this case, the red spots are on a particularly curvy location of the head .. In other cases this differences is less noticeable.

wmvanvliet, 2015-08-08:
I think what you need here is a topo plot with a "skirt". The EEGLAB plot is cleaner, but the positions of the electrodes are wrong (unless you used a very strange cap). @dengemann Can MNE plot topo with a skirt?

agramfort, 2015-08-08:
would this https://github.com/scot-dev/scot/pull/60 help?

wmvanvliet, 2015-08-08:
If you look at @ImmanuelSamuel's MNE Python image, you see that electrodes on the side of the head are all squished together. The topo plot is simply a top-down view. If you look at @ImmanuelSamuel's EEGLAB image, you see that these electrodes are "spread out", making the image much clearer. However, now the electrodes don't align with the head circle anymore (it's like the subject is wearing a [Kippah](https://en.wikipedia.org/wiki/Kippah)).

A solution to this is to scale the head circle so that the electrodes on the side of the head fall *outside* of the head circle. In standard 10-5 positioning, Oz should be *on* the head circle and Nz should be *outside*.

We should definitely consider borrowing from the above mentioned pull request as well.

I will look into it on Monday.

jona-sassenhagen, 2015-08-08:
@wmvanvliet while you're at it: one thing I like about the EEGLAB way is that electrodes are plotted outside the head circle if it makes sense to do so. e.g., mastoids, suborbital electrodes, and OZ actually end up outside the head circle.

wmvanvliet, 2015-08-10:
@jona-sassenhagen yes, that's what I mean with the "skirt"

wmvanvliet, 2015-08-10:
I looked into the matter. MNE and EEGLAB approach mapping the top half of a sphere to a plane differently.

Electrode coordinates are given in spherical coordinates: radius, inclination and azimuth.
 
MNE uses a top-side view of the sphere, inclination is simply ignored: `polar_to_cart(radius, azimuth)`. This means electrodes near the edges get squashed together.

EEGLAB calculates the arc length between the top of the sphere (Cz) and each electrode. This is used as substitute for the radius: `polar_to_cart(arc_length, azimuth)`. This means electrodes near the edges appear to be stretched outwards. This is used by FieldTrip as well. This screenshot shows it off nicely: http://sccn.ucsd.edu/eeglab/allfunctions/topoplot.html

Lets have a discussion first on what we prefer. @dengemann @haribharadwaj @christianbrodbeck?

wmvanvliet, 2015-08-10:
EEGLAB's approach reveals more detail near the edges, however MNE's current approach distorts less.

Obviously the best way to visualize a sphere is as an actual 3D sphere... :)

Eric89GXL, 2015-08-10:
Is that the same as a conic projection? If so, then we can just have the default be `projection='azimuthal'` (current behavior) and add `projection='conic'`. It seems like it would only be a few lines.

wmvanvliet, 2015-08-10:
Yes it is! Now I'm off to read the wikipedia page...

Eric89GXL, 2015-08-10:
A fun challenge for @jaeilepp or someone else would be to add turntable-like interactive click-drag moving for the `azimuthal` plots so that different angles of the sphere can be shown :) But I'm not sure it would be used often enough to justify the effort.

wmvanvliet, 2015-08-10:
Of course, with azimuthal projection, points below the equator (like mastoid reference electrodes) are problematic. This is the case in @ImmanuelSamuel's Biosemi128 montage and if you look at the screenshot you see that near the bottom edge there are twice the number of electrodes there should be, which screws up the data interpolation.

wmvanvliet, 2015-08-10:
I suggest to keep 'azimuthal' as default, but raise a warning if electrodes below the equator are present. The warning suggests the user to use conic projection instead.

@jona-sassenhagen Conic projection allows straightforwardly for a "skirt".

agramfort, 2015-08-10:
sounds like a plan.

jona-sassenhagen, 2015-08-10:
Conic projection would be cool.

wmvanvliet, 2015-08-10:
Wait... `plot_topomap` doesn't map 3D to 2D at all. It expects a layout object. All the mapping code is in the layout module.

@ImmanuelSamuel the `pos` parameter of `plot_topomap` expects 2D coordinates. Unfortunately it silently picks the first two coordinates without raising a warning or error. So in your case, it simply picks the x and y coordinates and drops the z (which in this case results in a azimuthal projection). You should first generate a layout object (`mne.channels.make_eeg_layout`) for your EEG channels and use the `layout` parameter.

I will add the conic projection option to the automatic layout generation code, where it belongs.

wmvanvliet, 2015-08-10:
Conic projection without skirt:

[image]

Starting to look like the EEGLAB output... I'll open a PR as soon as this is ready.

agramfort, 2015-08-10:
beautiful. At the same time can you raise an error if pos has more than 2
columns in plot_topomap ? thx heaps

ImmanuelSamuel, 2015-08-10:
Thanks @wmvanvliet and everyone for this. Yeah I started using plot_topomap by creating an evoked object but that didn't have the axis parameters for me to put in a subplot. So this is great!!

wmvanvliet, 2015-08-12:
See #2399

wmvanvliet, 2015-08-14:
@ImmanuelSamuel The improved topomap code was just merged. Check this gist for how to plot your data:

https://gist.github.com/wmvanvliet/6d7c78ea329d4e9e1217

closed 2015-08-14 by wmvanvliet

ImmanuelSamuel, 2015-09-12:
Thanks! Can you also add outlnes='skirt' for mne.viz.plot_topomap()? evoked.plot_topomap has a title that shows the time which I dont want.

agramfort, 2015-09-13:
can you give it a try yourself?

let us know if you need help.

wmvanvliet, 2015-09-13:
it already has that option. All the various topomap plotting functions have it. Please give it a try and let is know if you need help.

ImmanuelSamuel, 2015-09-13:
I did and it doesn't seem to work that is why I thought it wasn't implemented. But then again it didn't raise an error for 'skirt' option sorry I missed that.

[image]

wmvanvliet, 2015-09-13:
you cannot give `montage.pos` as electrode positions to the `plot_topomap` function. It doesn't complain about this (I'll make a PR so you get an error when you try this), but the electrode positions will be wrong.
What you have to do first is make an instance of  `Layout`:

    temp = evoked.data
    l = mne.channels.make_eeg_layout(evoked.info)
    mne.viz.plot_topomap(temp[:, 0], l.pos[:, :2], outlines='skirt')

ImmanuelSamuel, 2015-09-13:
Ah I see. Thank you.