mne-tools/mne-python #2391: drop_epochs and then drop_bad_epochs not dropping

mattjerhart, 2015-08-07:
```python
#...
epochs[c].drop_epochs(drop_ix,reason='balance for matched stim')  # says it dropped what I wanted
epochs.drop_bad_epochs() #says dropped 0
```

Ultimately, those epochs are all still in epochs. Am I doing something wrong?

Eric89GXL, 2015-08-07:
When you do `epochs[c]`, you are actually making a new `Epochs` instance, so any changes you make to that instance will have no effect on the original `epochs` object. If you do:
```
epochs_2 = epochs[c]
epochs_2.drop_epochs(...)
epochs_2.drop_bad_epochs()
```
You should see `epochs2` has the required epochs removed, while `epochs` in unchanged.

mattjerhart, 2015-08-07:
Got it, is this how I should be dropping epochs for a specific condition or is there a cleaner way? c = 'cond_name' in this case, sorry, didn't realize c could be integers of trial numbers as well.

Eric89GXL, 2015-08-07:
I usually figure it out by doing the arithmetic with indices into the `events` array. @agramfort might know a better way.

If you are trying to equalize trial counts, we do provide some functionality for doing that in a principled way, too -- look for `equalize_epoch_counts`.

mattjerhart, 2015-08-07:
I working with a word repetition task, so I'm dropping words that aren't in both conditions because of previous rejection. equalize_epoch_counts is very nice, but I've got to pick out some specific trials.

agramfort, 2015-08-07:
see

epochs.selection

to know what epochs remain after calling drop_bad_epochs()

Eric89GXL, 2015-08-07:
Instead of dropping ones you don't want, maybe select the ones you do want, as:
```
epochs_2 = epochs[['cond_a', 'cond_b', ...]]
```
?

mattjerhart, 2015-08-07:
Thanks for the help. This seems to be doing what I want:

```python
epoch_tmp = [];
words_in_both = np.intersect1d(events[epochs['first'].selection,3], events[epochs['2sec'].selection,3]); #col3 is unique word ids
for c in ['first','2sec']:
    drop_ix = np.logical_not(np.in1d(events[epochs[c].selection,3],words_in_both))
    epoch_tmp.append(epochs[c])
    epoch_tmp[-1] .drop_epochs(drop_ix,reason='balance for matched stim')  # says it dropped what I wanted
    epoch_tmp[-1] .drop_bad_epochs() #says dropped 0
epochs = []
epochs = mne.epochs.concatenate_epochs(epoch_tmp) 
```

closed 2015-08-07 by mattjerhart