mne-tools/mne-python #2393: Q: help on 'too many neighbors bug'

kingjr, 2015-08-09:
Hey,

I'm encountering an error on some of my subjects when running mne.bem.make_watershed_bem and I'm not sure what to do here. Do you think it's related to a non-closed mesh? 

```
make_watershed_bem(subject='subject05', subjects_dir='data',
                   overwrite=True, volume='T1', atlas=False,
                   gcaatlas=False, preflood=None)
```

leads to
```
3. Creating medium tessellation...
3.1 Decimating the dense tessellation...
3.2 Creating data/subjects/subject05/bem/subject05-head-medium.fif
Running subprocess: mne_surf2bem --surf /tmp/tmpU6HNDB/tmp-surf.surf --id 4 --fif data/subjects/subject05/bem/subject05-head-medium.fif
stderr:

mne_surf2bem version 1.8 compiled at Aug  9 2015 04:22:26

input  file #   1 : /tmp/tmpU6HNDB/tmp-surf.surf / id = 4 / sigma N/A
output file       : data/subjects/subject05/bem/subject05-head-medium.fif

Triangle file :  nvert = 14703 ntri = 29998
	Triangle and vertex normals and neighboring triangles...[done]
	Vertex neighbors...Too many neighbors for vertex 10884.

(u'', u'\nmne_surf2bem version 1.8 compiled at Aug  9 2015 04:22:26\n\ninput  file #   1 : /tmp/tmpU6HNDB/tmp-surf.surf / id = 4 / sigma N/A\noutput file       : data/subjects/subject05/bem/subject05-head-medium.fif\n\nTriangle file :  nvert = 14703 ntri = 29998\n\tTriangle and vertex normals and neighboring triangles...[done]\n\tVertex neighbors...Too many neighbors for vertex 10884.\n')
```

which ends up crashing 

```
---------------------------------------------------------------------------
CalledProcessError                        Traceback (most recent call last)
scripts/run_mne_anatomy.py in <module>()
     44     # Make scalp surfaces
     45     make_scalp_surfaces(subjects_dir, subject, force='store_true',
---> 46                         overwrite='store_true', verbose=None)
     47 
     48     # Make morphs to fsaverage

/home/jrking/anaconda/lib/python2.7/site-packages/mne/commands/mne_make_scalp_surfaces.pyc in _run(subjects_dir, subject, force, overwrite, verbose)

/home/jrking/anaconda/lib/python2.7/site-packages/mne/utils.pyc in verbose(function, *args, **kwargs)
    538         finally:
    539             set_log_level(old_level)
--> 540     return function(*args, **kwargs)
    541 
    542 

/home/jrking/anaconda/lib/python2.7/site-packages/mne/commands/mne_make_scalp_surfaces.pyc in _run(subjects_dir, subject, force, overwrite, verbose)
    136         try:
    137             run_subprocess(['mne_surf2bem', '--surf', surf_fname, '--id', '4',
--> 138                             '--fif', other_fname], env=this_env)
    139         finally:
    140             del tempdir

/home/jrking/anaconda/lib/python2.7/site-packages/mne/utils.pyc in run_subprocess(command, verbose, *args, **kwargs)

/home/jrking/anaconda/lib/python2.7/site-packages/mne/utils.pyc in verbose(function, *args, **kwargs)
    538         finally:
    539             set_log_level(old_level)
--> 540     return function(*args, **kwargs)
    541 
    542 

/home/jrking/anaconda/lib/python2.7/site-packages/mne/utils.pyc in run_subprocess(command, verbose, *args, **kwargs)
    806         err_fun = subprocess.CalledProcessError.__init__
    807         if 'output' in inspect.getargspec(err_fun).args:
--> 808             raise subprocess.CalledProcessError(p.returncode, command, output)
    809         else:
    810             raise subprocess.CalledProcessError(p.returncode, command)

CalledProcessError: Command '['mne_surf2bem', '--surf', '/tmp/tmpU6HNDB/tmp-surf.surf', '--id', '4', '--fif', 'data/subjects/subject05/bem/subject05-head-medium.fif']' returned non-zero exit status 1

```

Thanks a lot!

PS: shall I use the mailing list rather than github for this kind of questions?

agramfort, 2015-08-09:
yes use the mailing list.

have you tried to use the atlas or gcaatlas options which help if you have
a crappy MRI.

Eric89GXL, 2015-08-09:
I usually get that error when there is noise at the bottom of the MRI. 99%
of the time, zeroing out the bottom slice in T1.mgz and rerunning the
surface generation fixes it for me.

kingjr, 2015-08-09:
@Eric89GXL Thanks, I never manipulated mgz data, do you have a script that check this noise and zero it?

Eric89GXL, 2015-08-09:
No, we do it manually using `FreeView` because it's so rare, but come to think of it, it should be almost trivial with `nibabel` to do the zeroing. We never check the noise, we just try the zeroing fix and it almost always works.

kingjr, 2015-08-10:
It was a libgfortran lib issue missing from the os environment.

Thanks for your help!

closed 2015-08-10 by kingjr