mne-tools/mne-python #2397: BUG: GAT with LeaveOneLabelOut

kingjr, 2015-08-11:
@dengemann @agramfort 

The `GAT` doesn't work with the `LeaveOneLabelOut` cross val object because it doesn't have a `n_folds` params, which is used to check that identical data are passed to `fit` and `score` functions.

I'm not sure whether 
- I should ask the on the sklearn rep whether `n_folds` shouldn't be added to the LOLO class (after all, it makes sense to speak of folds for a LOLO doesn't it?)
- or whether I should just change the mne code to deal with this.

agramfort, 2015-08-12:
use an aux function to get the number of folds.

 - first try n_folds
- if not available use len(list(cv))

agramfort, 2015-08-12:
as usual PR welcome

closed 2015-08-12 by agramfort

kingjr, 2015-08-12:
I'll do the aux function.

> as usual PR welcome

You mean on sklearn?

agramfort, 2015-08-13:
no in time_gen.py in MNE

agramfort, 2015-08-13:
wait a second. It seems you can just do len(cv)

kingjr, 2015-08-13:
indeed, thanks. I'll wait for review in https://github.com/mne-tools/mne-python/pull/2401 to put everything in the same PR as this (and other things) are tiny changes.

agramfort, 2015-08-13:
I would make a dedicated PR as merging the other branch will require a bigger review...

kingjr, 2015-08-13:
ok