mne-tools/mne-python #2406: BUG: EpochsArray 'no matching events'

kingjr, 2015-08-15:
I think there may be a numerical error when creating epochs with `EpochsArray`?

```
# prepare
import numpy as np
from mne.epochs import EpochsArray
from mne.io.meas_info import create_info
ntrial, nchan, ntime = 100, 20, 10
data = np.ones((ntrial, nchan, ntime))
chan_types = ['eeg'] * nchan
chan_names = ['eeg_%02i' % chan for chan in range(nchan)]
events = np.zeros((ntrial, 3))
np.random.seed(0)
events[:, 2] = np.random.randn(1, ntrial)
info = create_info(chan_names, 256, chan_types)
```
```
# this works
events[:, 2] = np.round(1e7 * np.random.randn(1, ntrial))
EpochsArray(data, info, events=events, verbose=False)
```
```
# this doesn't
events[:, 2] = np.random.randn(1, ntrial)
EpochsArray(data, info, events=events, verbose=False)
```
```
> ValueError: No matching events found for 1.02017271172 (event id 1)
```

kingjr, 2015-08-15:
(I deal with continuous regressors coded in the events values, e.g. angle orientation, pich etc)

kingjr, 2015-08-15:
Not a numerical error, it just won't accept float. Any rational for this?

Eric89GXL, 2015-08-15:
Some reasons are historical (e.g., trigger values are integer coded in most MEG / EEG systems). But one good reason is that float comparisons are notoriously flakey; if actual `==` / exact functionality is desired, using integers and integer arithmetic is much safer. Is it possible for you to convert your scheme to an integer-based representation?

Eric89GXL, 2015-08-15:
In any case, at the very least we should raise an error for now if non-integer event codes are supplied.

kingjr, 2015-08-15:
> Is it possible for you to convert your scheme to an integer-based representation?

I deal with angles in radians throughout the analyses but there's always a way. But this may be a problem faced by others, so +1 for the raise.

Eric89GXL, 2015-08-16:
It might be slightly annoying, but you could construct a `dict` to map the floats to ints.

I'm going to reopen this issue until we take care of the error-raising with PR. I can take care of it Monday.

kingjr, 2015-08-16:
> It might be slightly annoying, but you could construct a dict to map the floats to ints.

Maybe this will work in most cases indeed. In my (highly ad hoc) case this isn't the case: I construct meta epochs (I have ~ 6000 thousands trials, which is often too much for my laptop),  by averaging a variable number of epochs, and linking it to the mean event value (which thus vary as a function of the meta epoch parameter). So I can't predict the corresponding dict easily - nor can I just map all this to integers, as I couldn't apply cos and sin functions to them afterwards. TMI? maybe.

agramfort, 2015-08-16:
@kingjr can you take care of the PR that raises a warning when passing non
integer values as events in EpochsArray?

Eric89GXL, 2015-08-16:
I think it needs to be an error to be safe

agramfort, 2015-08-16:
Fine with me

PR #2415 (2015-08-18): MRG: Enforce ints for events
Fixes the six bugs we had not making `events` integers. Builds on #2411. Closes #2406.
closed 2015-08-16 by kingjr