mne-tools/mne-python #2408: BUG? Time after decimation & EpochsArray

kingjr, 2015-08-16:
@Eric89GXL Decimating epochs and recreating new epochs with `EpochsArray`, don't generate identical `times`.

```
# prepare
import numpy as np
from mne.io.meas_info import create_info
from mne.epochs import EpochsArray
sfreq = 250
n_trial, n_chans, n_time = 100, 32, 63
data = np.random.randn(n_trial, n_chans, n_time)
chan_names = ['eeg' + str(ii) for ii in range(32)]
chan_types = ['eeg' for ii in range(32)]
info = create_info(chan_names, sfreq, chan_types)
events = np.c_[np.cumsum(np.ones(n_trial)) * 5 * sfreq,
               np.zeros(n_trial), np.zeros(n_trial)]
epochs = EpochsArray(data, info, events=events, verbose=False)
```

```
epochs.decimate(3)
print([epochs.times[0], epochs.times[-1]])
epochs = EpochsArray(epochs._data, epochs.info, events=events, verbose=False)
print([epochs.times[0], epochs.times[-1]])
```

gives

```
[0.0, 0.012, 0.23999999999999999]
[0.0, 0.012, 0.24000000000000002]
```

Eric89GXL, 2015-08-16:
Floating point arithmetic is not exact, I'd expect errors of that order of
magnitude unfortunately

agramfort, 2015-08-16:
indeed. Difference is 2.7755575615628914e-17 ... beyond float64 eps.

@kingjr welcome to floating point arithmetic ...

kingjr, 2015-08-16:
Ok so this makes GAT crash because the size of the time slice used in the test set (based on a new epoch) is smaller than the one used in the training. Shall I just change GAT function to check for `almost_equal`?

agramfort, 2015-08-16:
Yes

kingjr, 2015-08-16:
So... I went for correcting the GAT, but no success there. The issue can become larger than these 1e-17 errors:

```
import numpy as np
from mne.io.meas_info import create_info
from mne.epochs import EpochsArray
sfreq = 20
data = np.random.randn(1, 1, 20)
chan_names = chan_types = ['eeg']
info = create_info(chan_names, sfreq, chan_types)
events = np.zeros((1, 3))
epochs = EpochsArray(data, info, events=events, verbose=False, tmin=-.200)
epochs.decimate(3)
print([epochs.times[ii] for ii in [0, 1, -1]])
epochs = EpochsArray(epochs._data, epochs.info, events=events, verbose=False, tmin=epochs.times[0])
print([epochs.times[ii] for ii in [0, 1, -1]])
```

returns
```
[-0.20000000000000001, -0.050000000000000003, 0.69999999999999996]
[-0.14999999999999999, 0.0, 0.75]
```

kingjr, 2015-08-16:
but maybe that's a separate issue on the t0 is defined from tmin?

Eric89GXL, 2015-08-16:
Yeah, decimation will keep the point `t=0` @ 0

kingjr, 2015-08-16:
> Yeah, decimation will keep the point t=0 @ 0

Ok but why the discrepancy between `epochs` and the `new_epochs`?

Eric89GXL, 2015-08-16:
Oh that is weird... not sure what's going on there exactly

kingjr, 2015-08-18:
Can we reopen this issue so as not to forget about it?

agramfort, 2015-08-18:
what's the way to move forward?

Eric89GXL, 2015-09-03:
This bug should be fixed on latest `master`, I ran it and got:
```
[-0.20000000000000001, -0.050000000000000003, 0.69999999999999996]
[-0.20000000000000001, -0.050000000000000003, 0.69999999999999996]
```

closed 2015-09-03 by Eric89GXL