mne-tools/mne-python #2409: XDAWN example: cross validation?

kingjr, 2015-08-16:
@alexandrebarachant I think we should add a cross validation step in the XDAWN example, there could be an overfitting here couldn't there?

kingjr, 2015-08-16:
@trachelr This is also applicable to your plot_pattern example?

agramfort, 2015-08-16:
what examples?

kingjr, 2015-08-16:
These two:

http://martinos.org/mne/dev/auto_examples/decoding/plot_linear_model_patterns.html#sphx-glr-auto-examples-decoding-plot-linear-model-patterns-py

http://martinos.org/mne/dev/auto_examples/preprocessing/plot_xdawn_denoising.html#sphx-glr-auto-examples-preprocessing-plot-xdawn-denoising-py

agramfort, 2015-08-17:
we don't print / compute a prediction score here. So we keep the examples short and more readable.

I am closing. Reopen if you disagree.

closed 2015-08-17 by agramfort

kingjr, 2015-08-18:
You plot the data fitted with itself. It's a double dip.

kingjr, 2015-08-18:
(I can't reopen)

kingjr, 2015-08-18:
If you want to keep the examples simple, you could just train on the first half of the data, and apply XDawn or whatever onto the second half of the epochs. I think we need to make people aware of these issue right from the beginning, if we don't want to be spotted by future [~~anti-voodoo~~ puzzled statisticians](http://pps.sagepub.com/content/4/3/274.short?rss=1&ssource=mfc).

agramfort, 2015-08-18:
just provoking, if you show me the ERF by averaging 100% of the data is
this a double dip? Basically I am just saying that it depends of what you
claim and what you report in terms for stats.

alexandrebarachant, 2015-08-18:
I would say your are not wrong. 
You should not apply any stats analysis on top of Xdawn if it wasn't properly cross-validated. However, for a pure visualization point of view, there is nothing wrong in the example.

On a larger basis, one could argue that the same problem arise with any preprocessing step (ICA, PCA etc) in the way that you generally select components that contain your signal of interest.

kingjr, 2015-08-18:
@agramfort Averaging is done in a single step so it's not a double dip. Classic stats (std, confidence interval etc) assuming the way you average is valid with regard to your stats assumption (mainly normality). On the contrary these cleaning method 1) fit and 2) apply a transformation in an informed way  (i.e. they know trials' category), hence the double dip.

No one will use XDawn just for viz; they will use it as a cleaning method in their pipeline. And if they used Xdawn on a particular regressor of interest, I'd be surprise to find anyone not looking at this regressor of interest later on.

@alexandrebarachant I think the potentially important difference with xdawn or the Romain's patterns' on the one hand, and PCA or ICA on the other hand, is that the former are supervised whereas the latter aren't. Intuitively, I would think that double dipping issues are more critical in supervised cases - but perhaps I'm wrong? 

Overall, imagine we were to do this with decoding: fit your data on all epochs, and show single trial predictions with no CV. Formally it's the same steps as XDawn, just in a different space, isn't it?

agramfort, 2015-08-19:
ok point taken. But again it depends on what you claim. Would you send a PR
adding a warning note in both of these examples?

kingjr, 2015-08-19:
Sure. I think the simplest would be something like
```
# To avoid circular analysis (a.k.a 'double dipping'), the model is fitted and applied
# on independent sets of epochs.
cov = compute_covariance(epochs[:n])
xd.fit(epochs[:n], signal_cov=cov)
clean_epochs = xd.apply(epochs[n:])
```

The alternative are ii) using a sklearn cv (which I think add to much complexity in the example) or iii) just keep the current example and throw a warning at the end (which I think is a bit like: we're guiding you to the wrong track but we're warning you)

agramfort, 2015-08-19:
does it visually affect the result?

kingjr, 2015-08-19:
no

agramfort, 2015-08-20:
good. PR more than welcome.