mne-tools/mne-python #2416: ENH Website

teonlamont, 2015-08-17:
The website looks great. I just noticed a few things when I was browsing. Feel free to address as you see fit.

In mobile browsing:
- [ ] there's an overlap in the widget containing the number of lines of code and the documentation section
- [ ] the university logos are off-centered with the package logo
- [ ] red bar about dev version is hidden by the toolbar

In general:
- [ ] missing sections from reference: ``read_dig_montage`` and ``linear_regression_raw``.

Eric89GXL, 2015-08-17:
I actually don't think they're off-centered, I think it's a visual thing due to the fact that the logos aren't rectangular. At least according to the HTML they should be centered

Eric89GXL, 2015-08-17:
I don't see the hiding or the overlap, either, this is what mine looks like:

[image]

[image]

What browser are you using?

teonlamont, 2015-08-17:
@Eric89GXL I meant browsing on iPhone and other mobile devices. I think it's just how bootstrap is set up to the wrap.

teonlamont, 2015-08-17:
[image]

[image]

Eric89GXL, 2015-08-17:
Ahh, sorry, missed the mobile bit

agramfort, 2015-08-18:
reference fixes are more than welcome

Eric89GXL, 2015-10-19:
Fixed by https://github.com/mne-tools/mne-python/commit/632e49f0470fc9526936dbb474fd6aa46501fe4d. I didn't fix the logo centering, though, because I didn't see an easy way to do it, and it's a minor point. If someone else knows how or wants to do it, feel free.

closed 2015-10-19 by Eric89GXL