mne-tools/mne-python #2419: Easily append zeros to a Raw object

choldgraf, 2015-08-18:
What's the easiest way to quickly row the length of a Raw object? I've tried using a RawArray and `.append`, but this throws an error because the types of the two are not the same. Basically, one of my datasets is a prime number in length so it's taking forever to do a Hilbert transform, so I'd like to just append zeros to get it faster.

Eric89GXL, 2015-08-18:
Maybe we should just add an option to the hilbert transform, instead?

choldgraf, 2015-08-18:
Also an option, yeah. In scipy you can specify "N" and it'll append zeros to that point. Could just allow Hilbert to take "N" and then it'd supply that to the call to "Hilbert" and then take the first `n_times` points in the output of Hilbert.

choldgraf, 2015-08-18:
Actually there are probably other places where this would be useful too. I think we've spoken about this before, but any time that it's possible to pad for efficiency purposes I think is a bit clunky right now. E.g., right now there are functions that let you supply a padding, but the padding is symmetric, so if you're padding in order to get to a specific length and your signal is an odd number, you have to figure out a way to add / subtract 1 datapoint.

Eric89GXL, 2015-08-18:
Which functions are you thinking of? The FIR filtering functions are already written in a way to optimize the FFT length, so you shouldn't need to worry about it there. If the TFR functions are not that way, then maybe we should rewrite them to make use of the FIR code, which has been a bigger target of optimization.

choldgraf, 2015-08-18:
TFR is the main situation where I've run into that issue, ya.

Either way, right now I'm doing something like:

```python
# Load data
ecog = mne.io.Raw(data_path + '{0}/clean/{0}_ecog_clean_raw.fif'.format(s_name), preload=True, add_eeg_ref=False)
ecog = mne.io.RawArray(ecog._data, ecog.info)

# Make length a pow of 2
orig_times = ecog.n_times
new_len = 2**np.ceil(np.log2(orig_times))
diff = new_len - ecog.n_times
zeros = mne.io.RawArray(np.zeros([ecog._data.shape[0], diff]), info=ecog.info)
ecog.append(zeros)

# Filter for HG and extract amplitude
ecog.filter(70, 150, n_jobs=4)
ecog.apply_hilbert(range(len(ecog.ch_names)), envelope=True, n_jobs=4)
ecog = mne.io.RawArray(ecog._data[:, :orig_times], ecog.info)
```

which is pretty hacky :)

Eric89GXL, 2015-08-18:
Yeah, those are the sorts of hacks I'd prefer to avoid by instead improving the filtering code where necessary :)

closed 2015-12-21 by choldgraf