mne-tools/mne-python #2420: Spinning off the _hdf5 module?

choldgraf, 2015-08-19:
There was just a discussion on the scikit learn listserv where the _hdf5 module in MNE came up. I feel like we've chatted about this here and there in the past, but wanted to open it up more formally.

That _hdf5 module is pretty darn useful, and I think it's too bad that it currently exists hidden to the user. Does anyone think it would be better if that was either a module that you could find more easily (e.g., mne.io.hdf5), or even spun off into its own mini package or something?

dengemann, 2015-08-19:
@Eric89GXL agreed on releasing it as a separate package as we need this functionality outside of MNE. This would also help reducing maintenance efforts. Now we just have to do it.

agramfort, 2015-08-19:
I would still keep a copy in mne to avoid another dependency.

dengemann, 2015-08-19:
The idea was to move it to externals.

agramfort, 2015-08-19:
Fine with me

Eric89GXL, 2015-08-19:
What should we call it? If we can come up with a name, I can start a repo and make it in an hour or two.

Eric89GXL, 2015-08-19:
I propose `pooh`, because it's a funny name (pooh bear) is the spirit of Python, and can stand for Python Objects Onto HDF5

`from pooh import write_hdf5, read_hdf5`

Eric89GXL, 2015-08-19:
cc @drammock @rkmaddox we need to choose a name to spin off our generic HDF5 writing routines, speak now or forever hold your peace

choldgraf, 2015-08-19:
lol, I really like pooh for pooh bear, though I also like it because of the
15 year old boy in me, which worries me for that particular name :) maybe
it's fine though? It is catchy...

drammock, 2015-08-19:
what about a riff on "high five", like `hifive` or similar?

Eric89GXL, 2015-08-19:
I like it, but it conflicts with at least one existing name:

https://pypi.python.org/pypi/hifive/1.0.3

drammock, 2015-08-19:
Dang. Other ideas: `pyfive`, `reedwright`

choldgraf, 2015-08-19:
I like pyfive...it sounds like a startup that lets you order high-fives
with your smartphone

Eric89GXL, 2015-08-19:
I like pyfive (not really reedwright), but again:

https://testpypi.python.org/pypi/PyFive/1.0a1

drammock, 2015-08-19:
dammit.  well, `pooh` then.  Or `eeyore`.  I can't think of a pithy name that conveys heirarchical organization, data, and snakes all at once.  Except maybe `IRS`.

choldgraf, 2015-08-19:
lol, bravo Daniel McCloy, that got a good chuckle out of me

dengemann, 2015-08-19:
seriously, why not one of these:

- h5io
- h5ext
- h5api
- pyio
- pydump
- h5pickle

it should a) somehow refer to h5py but point out that it's an extensnion or
API or b) refer to serialization

Eric89GXL, 2015-08-19:
FYI @drammock Python actually refers to Monty Python (not the reptile), but I see your point :)

If only @choldgraf and I like `pooh`, then I can probably live with `h5ext` or `h5io`. `hfive` is also available.

@agramfort you're always good for an opinion, WDYT?

agramfort, 2015-08-19:
I would go for h5io as it's explicit

choldgraf, 2015-08-19:
I like h5io, though:

It is a bit redundant...I mean, the whole point of HDF5 is io, right? Just
calling it h5io makes it sound like it could be an *alternative* to h5py,
not an extension...

Eric89GXL, 2015-08-19:
I agree it's a bit redundant, but it seems like the best non-`pooh` option

choldgraf, 2015-08-19:
fair enough, I like it if people don't think it'll be confusing

Eric89GXL, 2015-08-19:
Done, now we just need to incorporate it:

https://github.com/h5io/h5io

agramfort, 2015-08-20:
let's move this to externals then now.

Eric89GXL, 2015-08-20:
I'm on it

PR #2424 (2015-08-21): MRG: Use h5io and fix tests
Closes #2420.

I also included some fixes I needed to get the test suite to run on my machine without as many warnings.
closed 2015-08-21 by agramfort