mne-tools/mne-python #2425: Connectivity and surface plotting for Agglomerative Clustering in scikit

mattjerhart, 2015-08-20:
I was hoping to try out some agglomerative clustering on a group average stc file as it seemed pretty easy. I was expecting contiguous clusters but I'm getting getting disconnect spots all over in some labels. @agramfort is actually on some of the scikit examples, but I mainly just want to make sure I'm using mne correctly, especially with regard to extracting the connectivity and creating/plotting a source object.

```python
from sklearn import cluster
from mne import spatial_tris_connectivity, grade_to_tris
connectivity = spatial_tris_connectivity(grade_to_tris(4),1) # is this the right? There's dist and src connectivity too.
k = 100;
model = cluster.FeatureAgglomeration(n_clusters=k,connectivity=connectivity,linkage='average',affinity='cityblock')
fit = model.fit(mean_stc.data.T)

an_roi = np.zeros(fit.labels_.shape)
an_roi[fit.labels_==0] = 5# now look at one cluster on the surface
cluster_stc = mne.SourceEstimate(an_roi[:,None],vertices=mean_stc.vertices,tmin=1,tstep=1) #is this an ok way to visualize such a cluster on the surface?

brain = cluster_stc.plot(subject='fsaverage',hemi='split',colormap="jet",
              subjects_dir=fs_dir,views=['lat', 'med'],
              config_opts=dict(width=2500, height=1400))
```

agramfort, 2015-08-21:
please share a full script as gist with data to reproduce.

mattjerhart, 2015-08-21:
[Here's a gist that I actually was able to get working, thought there are enormous differences in cluster size.](https://gist.github.com/mattjerhart/ac4556330ff262f6ee5b). The issue I was having had to do with not morphing from grade 4 to 5 before clustering and plotting the surface. 

I also tried kmeans and that provided much nicer looking clusters, and produces surprisingly contiguous ROIs. It's too bad FeatureAgglomeration is so uneven because I like that it enforces spatial connectivity.   As a quick way to look at the data, I like kmeans ROIs best so far. Do you have any recommendations on how to create ROIs based on waveform similarity that, preferably, enforces spatial connectivity, results in ROIs reasonably similar in size, can automatically determine the number of clusters, and/or will run fairly quickly on stc sized data?

agramfort, 2015-08-23:
did you try using linkage='ward'

it should provide provide more even distributed clusters.

you might want to read this:

http://www.ncbi.nlm.nih.gov/pmc/articles/PMC4076743/

although it's for fMRI originally.

I am closing the issue which for the future i recommend posting on the mailing list. Github is really for bugs.

closed 2015-08-23 by agramfort